MaStr/batcontrol #1: Lastprofil

gjuecat, 2023-12-09:
Hallo Muexxl,

wie muss ich das Lastprofil aufbauen? Ist das die Leistung zum Timestamp oder die verbrauchte Energie der aktuellen Stunde?

Danke und lg

gjuecat, 2023-12-09:
Hat sich nach dem Schreiben eigentlich grad beantwortet - bleibt sich ja egal (wenns das 1 Std. Intervall ist).

closed 2023-12-09 by gjuecat

muexxl, 2023-12-11:
Absolut richtig. 
Ein Eintrag ist die verbrauchte Leistung in Wh für die entsprechende Stunde. 
Dies entspricht der equivalenten Last  in W für die gesamte Stunde.

muexxl, 2023-12-17:
Mal eine Nachfrage zu deinem Lastprofil: 
Wie hast du dein Lastprofil erzeugt? 
Inwiefern weicht es von dem Standardlastprofil ab?
Würdest du dein Lastprofil der Community zur Verfügung stellen?

gjuecat, 2023-12-24:
Hallo,
ich habe mein Lastprofil vorerst nur im Excel generiert und geschätzt (kenne meinen "Leerlauf" vom Haus und weiß wann normalerweise gekocht wird. Sonderverbraucher wie Pool, ... habe ich vorerst nicht berücksichtigt. 
Ich habe aber für mich die Solaranzeige [https://solaranzeige.de/phpBB3/solaranzeige.php](https://solaranzeige.de/phpBB3/solaranzeige.php) laufen, aus der habe ich mir mein "richtiges" Lastprofil auch exportiert. 
Inwiefern ein zur Verfügung stellen Sinn macht weiß ich somit nicht, kann ich aber gerne machen.

gjuecat, 2024-11-05:
Hi,
noch ne Frage: weekday = 0 ist eh der Montag?

MaStr, 2024-11-06:
Korrekt. Montag ist 0, Sonntag ist 6.

https://docs.python.org/3/library/datetime.html nach "weekday" suchen.