MaStr/batcontrol #2: Problem beim Start

gjuecat, 2023-12-10:
Hallo Muexxl,

ich habe meine Config und das Lastprofil soweit fertig und wollte nun das Programm starten. Leider kommen dabei aber einige Fehlermeldungen. Hast du eine Idee, was hier noch falsch ist?

```
docker@s36-debian-1:~/batcontrol/env_batcontrol$ source bin/activate
(env_batcontrol) docker@s36-debian-1:~/batcontrol/env_batcontrol$ python batcontrol.py
2023-12-10 11:43:12 INFO [FCConsumption] loading module
2023-12-10 11:43:12 INFO [FCSolar] loading module
2023-12-10 11:43:12 INFO [Fronius] loading module
2023-12-10 11:43:12 INFO [Main] Starting Batcontrol
Traceback (most recent call last):
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 413, in <module>
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 44, in __init__
    self.load_config(configfile)
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 94, in load_config
    config = yaml.safe_load(config_str)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/__init__.py", line 125, in safe_load
    return load(stream, SafeLoader)
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/__init__.py", line 81, in load
    return loader.get_single_data()
           ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/constructor.py", line 49, in get_single_data
    node = self.get_single_node()
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 36, in get_single_node
    document = self.compose_document()
               ^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 55, in compose_document
    node = self.compose_node(None, None)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 84, in compose_node
    node = self.compose_mapping_node(anchor)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 133, in compose_mapping_node
    item_value = self.compose_node(node, item_key)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/composer.py", line 64, in compose_node
    if self.check_event(AliasEvent):
       ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/parser.py", line 98, in check_event
    self.current_event = self.state()
                         ^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/parser.py", line 449, in parse_block_mapping_value
    if not self.check_token(KeyToken, ValueToken, BlockEndToken):
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/scanner.py", line 116, in check_token
    self.fetch_more_tokens()
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/scanner.py", line 235, in fetch_more_tokens
    return self.fetch_tag()
           ^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/scanner.py", line 632, in fetch_tag
    self.tokens.append(self.scan_tag())
                       ^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/lib/python3.11/site-packages/yaml/scanner.py", line 967, in scan_tag
    suffix = self.scan_tag_uri('tag', start_mark)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^

```

Danke und lg

gjuecat, 2023-12-10:
Hallo,

habe die Fehler nun etwas verkürzt (durch entfernete Sonderzeichen im Passwort), leider läufts aber immer noch nicht:

```
(env_batcontrol) docker@s36-debian-1:~/batcontrol/env_batcontrol$ python batcontrol.py
2023-12-10 19:03:36 INFO [FCConsumption] loading module
2023-12-10 19:03:36 INFO [FCSolar] loading module
2023-12-10 19:03:36 INFO [Fronius] loading module
2023-12-10 19:03:36 INFO [Main] Starting Batcontrol
Traceback (most recent call last):
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 413, in <module>
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 44, in __init__
    self.load_config(configfile)
  File "/home/docker/batcontrol/env_batcontrol/batcontrol.py", line 92, in load_config
    config_str = f.read()
                 ^^^^^^^^
  File "<frozen codecs>", line 322, in decode
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xa7 in position 680: invalid start byte
```

Danke und lg

mrhahn3000, 2023-12-10:
Hallo Jürgen,

das klingt für mich nach einem Datei Format Fehler deiner Config Datei.

mit welchem Editor hast du die erstellt/gespeichert. Hast du darauf geachtet, dass es im UTF8 Format ist? 

Scheint so, dass beim read der Config ein decoding error auftritt.

VG
Kevin

gjuecat, 2023-12-11:
Hallo Kevin,

das wars. Editiert habe ich mit PSPad, nach dem umstellen auf UTF-8 BOM konnte ich das Script starten.

lg
Jürgen

closed 2023-12-11 by gjuecat