MaStr/batcontrol #3: Wünsche zum LOG-File

gjuecat, 2023-12-24:
Hallo,
das LOG File wird mit der Zeit sehr groß. Wie kann ich die Größe beschränken bzw. ist es möglich die älteren Einträge ab einer bestimmten Dateigröße zu überschreiben?

Was ich auch super finden würde, wäre eine Möglichkeit die Ausgaben per Telegram Bot zu versenden (so wie es bei [Tesla Solar Charger](https://github.com/pkuehnel/TeslaSolarCharger) gemacht wird. Dann brauche ich nicht immer im Logfile  nach zu sehen und hätte die Infos auch am Handy (Benachrichtigungen am Handy würde ich dann auf stumm schalten).

lg
Jürgen

muexxl, 2024-02-09:
Hallo, 
ich habe jetzt v.0.1.5 rausgebracht. Dort gibt es die Möglichkeit die Größe der Logfiles zu reduzieren. Standard ist 100kB. 

Ein Telegrambot wäre sicherlich interessant. Dafür fehlt mit aber im Moment die Kapazität.

fnolting, 2024-02-19:
Sehr cooles Projekt - genau das, was ich gesucht habe! Ich habe ein Solaranzeige-Setup laufen und werde versuchen die Logs von batcontrol an das Script "meldungen_senden" (unterstützt Pushover, WhatsApp, Telegram & Signal) zu übergeben bzw. einen Meldungsblock zu bauen.

muexxl, 2024-02-19:
Freu mich, dass es dir gefällt und dass du darauf aufbauen willst. 
Wenn sich daraus irgendwelche Anforderungen ergeben, lass es mich wissen. Bisher ist das Logging eher zum debuggen und verstehen, was das Programm eigentlich so macht.

wiggal, 2024-04-25:
Hallo muexxl,
dein batcontrol ist sehr cool, aber auch sehr komplex.
Ich möchte wie andere auch gerne darauf aufbauen. 
Da die ganzen HTTPrequests sehr kompliziert sind, wäre es cool wenn du dein Wissen teilen könntest. Vielleicht kannst du eine kurze Doku über die Aufrufe (Login, Batteriemanagement usw.) schreiben, oder gibt es dazu irgendwo schon eine Beschreibung.
Noch schöner wäre eine Pythonmodul, die die verschieden Aufrufe ermöglicht.

Sorry für die vielen Anliegen
Gruß
Wiggal

muexxl, 2024-04-26:
Hallo Wiggal, 
schau dir mal das Python Modul "fronius" an. Das Modul musst du eigentlich nur einbinden. Dann kannst du dir eine Instanz der "FroniusWR" Klasse erstellen und damit den Wechselrichter kontrollieren. 
Erfüllt das deine Zwecke?
Viele Grüße, 
Müxxl

muexxl, 2024-04-26:
Ich schließe das Issue mal. 
Das Logfile-pruning wurde mit Commit  https://github.com/muexxl/batcontrol/commit/7b6ea3d4569b29297bb76c59dae55a6e33664d57 
eingeführt.

closed 2024-04-26 by muexxl