MaStr/batcontrol #4: Verständnisfrage zur Konfiguration von batcontrol

jobe39, 2024-01-15:
Den Grundgedanken, wie batcontrol funktionieren soll, habe ich verstanden. Die Konfiguration allerdings noch nicht ganz. Mein Wunsch wäre es ja, konfigurieren zu können, ab welchem Preis / KWh mein BYD Speicher geladen werden soll. In der YAML finde ich nur folgenden Entrag:

`min_price_difference: 0.05 # minimum price difference in Euro to justify charging your battery`

Wie ist der Eintrag zu verstehen? 0.05 Differenz zu was? Oder anders gefragt, was stelle ich hier ein, wenn ich bspw. ab 8 cent / kWh den Speicher befüllen möchte?

Der zweite Punkt betrifft das Lastprofil. Ich habe aktuell noch keines und wüßte auch nicht, wie ich zu einem kommen kann. Den Smartmeter gibts erst seit September 23, ebenso die PV Anlage mit Fronius Gen24 und BYD Speicher. Wie komme ich am besten zu einem Lastprofil und wozu wird dieses benötigt?

Vielen Dank schonmal vorab für die Erklärungen. Ich würde das Ganze gerne verstehen, bevor alles live umgesetzt wird.

mrhahn3000, 2024-01-19:
Hi Jobe, ich versuche mal aus der User Sicht deine Frage zu beantworten:

1. Die `min_price_difference` funktioniert wie folgt: Du verwendest einen dynamischen Stromanbieter, wie z.B. tibber. Jeden Tag ca. 13 Uhr werden die Strompreise pro Stunde bekannt gegeben. Du weißt also wann die niedrigsten und höchsten Preisstufen vorhanden sind auf eine 36h Zukunftsaussage. batcontrol erkennt nun diese Differenz innerhalb der 36h. Das Setting besagt nun also was deine minimale Differenz dieser Preisstufen sein muss, damit deine Batterie geladen werden soll. Es gibt in anderen Foren dazu eine Menge Theorie, ab wann das Sinn macht. Das was du möchtest (einen absoluten Preis festlegen) macht ja nur wenig Sinn, denn diese wird wahrscheinlich "nie" erreicht bzw. im relativen Verhältnis zum restlichen Tag muss geprüft werden, ob du nun deine Batterie einem zusätzlichen Ladezyklus aussetzt (Lebensdauer der Batterie nimmt pro Ladezyklus ab) oder ob du einfach den Strom direkt aus dem Netz konsumierst. Ab mindestens 5 Cent relativen Preisunterschied, macht es unter Berücksichtigung verschiedener wirtschaftlicher Betrachtungen Sinn die Batterie mit Netzstrom zu laden. Außerdem passiert die "Magic" nicht nur bei der Auswertung des Strompreises, sondern - Dank der Einbindung von FCSolar (forecast solar) eben auch Standort- und Dachausrichtungsspezifisch - die Auswertung der Sonneneinstrahlung und damit Ertragsprognose.

2. Lastprofil: Ich persönlich habe das mitgelieferte Lastprofil genommen und lediglich meinen Jahresverbrauch in der Config angegeben, da auch ich erst ca. 1 Jahr im Haus lebe und erst seit letztem Oktober die PV auf dem Dach habe. Das Lastprofil generell beschreibt pro Stunde eines Tages im Monat den Strombedarf. Also wie viele Wattstunden brauchst du am Tag aufgeteilt für jede Stunde. Diese Werte habe ich auch noch nicht und werden sich in der Regel um ein paar Wattstunden pro Haushalt unterscheiden (steile These, ich weiß :D) Du kannst (so werde ich es diese Jahr machen) anhand von Wechselrichterdaten/tibber-daten) deinen Bedarf in Home Assistant (o.Ä. Tools) tracken und aufzeichnen. Dann kannst du kontinuierlich dein Lastprofil "personalisieren". Wie gesagt, das macht am Ende aber nicht sooo viel aus.

Nimm es einfach mal "live". Stelle die Daten in der Config richtig ein und du wirst etwas bemerken. Allein die automatische Umschaltung, wann geladen, wann gehalten und wann entladen werden soll, finde ich persönlich auch ohne persönliches Lastprofil ungemein bereichernd für meine Anlage und hat mir effektiv schon den ein oder anderen Euro eingespart ;)

Cheers,
Kevin

jobe39, 2024-01-19:
Vielen Dank für die ausführliche Erklärung. Ich beschäftige mich noch nicht so lange mit der Thematik und hab da wohl ein wenig zu kurz gedacht. Die Solarprognose ist natürlich ein wichtiger Bestandteil, vor allem im Sommer. An Sonnentagen reicht der Akku aus, um die Nacht abzudecken und morgens startet dann ohnedies wieder die PV Einspeisung. Ich muß als erstes Mal abwarten, bis der Wechsel zu aWATTar durch ist (Tibber gibt es bei uns in AT nicht). Dann kommen noch Überlegungen zum E-Auto laden hinzu (zwei Stück), da bin ich mir noch nicht ganz sicher, wie ich das alles in Einklang miteinander bringe. Ich fürchte, wenn ich E-Auto laden + Speicher laden wegen günstigen Preises gleichzeitig anwerfe, könnte das für den Hausanschluß ein wenig viel werden. Smartes E-Auto laden wird über evcc durchgeführt werden, das läuft jetzt schon für PV Überschuß laden. Hier gibt es auch die Möglichkeit, das Entladen des Speichers für den Zeitraum des smarten Ladens zu sperren (sonst wandern die 10kW vom Hausspeicher ziemlich schnell ins Auto ;-) ). 

Ich werde das wohl mal Schritt für Schritt angehen:
* Abwarten bis der Wechsel zu aWATTar durch ist
* smartes E-Auto laden aktivieren
* Erfahrungen damit sammeln
* Schauen, wie ich battcontrol mit der bestehenden Umgebung in Einklang bringen kann

Falls Du zufällig schon Überlegungen hast, wie das smarte Laden eines E-Autos in diese Konfiguration eingebracht werden kann, bin ich für Tipss natürlich sehr dankbar.

lg,
  Gernot

muexxl, 2024-01-22:
@Kevin: Danke, super Antwort. Ich hätte es nicht besser beschreiben können!
@Gernot: 
Bisher habe ich kein E-Auto. Aber es macht natürlich total viel Sinn. 
1. Die Ladung des E-Autos zu berücksichtigen. 
2. Den Akku des E-Autos ggfs. über Bi-direktionales Laden zu nutzen. 

Bisher ist mir nicht ganz klar, wie man das Laden des E-Autos sinnvoll einbinden könnte. Aus Sicht von Batcontrol ist das E-Auto ein zunächst ein riesengroßer Verbraucher, der die Abweichung von der Verbrauchsprognose erhöht. 

Mit der passenden Wallbox könnte man ggfs. sogar den Ladezeitpunkt anpassen. E.g. Prio 1:  PV-Überschussladen, falls möglich/ vorhanden. Prio 2: Laden wenn es günstig ist.

Gernot, 2024-01-22:
Hi there

jobe39, 2024-01-22:
Im wesentlichen erledigt diese Arbeit evcc ([https://github.com/evcc-io/evcc](url) und das mittlerweile sehr gut. Bis hin zu: "Ich brauche morgen um 07:00 Uhr ein vollgeladenes Auto" - dann werden dann PV Überschuß und günstige Strompreise berücksichtigt. Da sehe ich bei uns im Moment den Hauptfokus. Die Frage ist dann eher, ob es Sinn macht evcc und battcontrol auch noch irgendwie in Einklang zu bringen. Das Auto muss ja nicht jeden Tag geladen werden. Auf der anderen Seite kann es vermutlich auch ziemlich schief gehen, wenn zwei Tools automatisiert den Speicher verwalten.

gjuecat, 2024-02-05:
@E-Auto laden:
falls wer einen Tesla über PV oder zu günstigen Strompreisen laden will, kann ich [https://github.com/pkuehnel/TeslaSolarCharger](url) bestens empfehlen. 

Eine Verbindung der beiden Tools Batcontrol und E-Auto laden macht ist meiner Meinung fast zwingend notwendig. Ich hatte das nämlich schon, dass in der Nacht zu günstigen Strompreisen, dann das Auto aus dem Speicher geladen wird => da wäre es dann hilfreich, wenn die Entnahme vom Speicher zumindest gesperrt wird. Bisher habe ich leider noch keine Idee, wie ich das anstellen könnte. Die letzten Wochen waren sich die Tools zum Glück einig und haben gleichzeitig den Energiebzug aus dem Netz angetriggert (was noch OK ist, wenn man die max. Leistung vom Hausanschluss nicht überschreitet).

jobe39, 2024-02-05:
Das Sperren der Batterie, wenn mit günstigem Strom das E-Auto geladen wird, macht [evcc](https://github.com/evcc-io/evcc) bereits. Sollte auch mit dem Tesla problemlos funktionieren.

gjuecat, 2024-02-05:
@jobe39 stell ich mir aber schwierig vor, wenn 2 Tools die Register im WR bzw. BYD beschreiben, was zu tun ist - optimalerweise sollten das eben abgestimmt sein.

muexxl, 2024-02-09:
Die ursprüngliche Frage wurde ja geklärt. Daher würde ich das Issue mal schließen.

closed 2024-02-09 by muexxl