MaStr/batcontrol #11: Anbindung an EVCC

MaStr, 2024-08-19:
# EVCC Plugin

This plugin connects batcontrol with the MQTT evcc API endpoint.

## Functions 

* Verify of evcc/status = online/offline
* Based on configuration subscribe to general or specific loadpoints
  * evcc/loadpoints/+/charging  -> true / false
  * evcc/loadpoints/1/charging  -> true / false

This information is brought into the evaluation of batcontrol about discharing is allowed or not.

## How does it work

* Separate evcc section in the configuration file.
* Configuration has it's own mqtt parameters. It does not use the MQTT API class of batcontrol.
* batcontrol receives an extra method, which is run through during 
  * is_discharge_allowed
  * set_wr_parameters - # discharge not allowed block.

If SOC  > always_allow_discharge_limit  ; allow discharge
if SOC  < always_allow_discharge_limit  ; disallow discharge

## Next steps
### Planned is optional via configuration file:
* send Battery SOC to EVCC
* introduce an additional operation mode in batcontrol "limited discharge".
  * Limit allowed discharge to a dynamic (expected usage by load profile) W
  * Limit allowed discharge by static value ( 500W via config )
     *  This requires a slightly different calculation during battery usage forecast and "recharge" evaluation.
  * set via configuration, which SOC% is relevant during EV-charing, this value can also be in sync with evcc

MaStr, 2024-10-24:
Im branch evcc-Connection habe ich eine erste Version fertig. Ich habe das Lokal gegen eine Testinstanz von evcc und batcontrol getestet.

WolfImBusch, 2024-10-26:
Hallo  @MaStr ,

super, super spannend, was Du und @muexxl da machen!!!!! Ich möchte Dich um Deine Hilfe bitten, da ich die Grundidee leider noch nicht ganz verstanden habe.

Kurz vorab: ich kenne und verwende evcc, das seit ein paar Wochen die Funktion bietet, in Abhängigkeit des Strompreises die Hausbatterie zu laden. batcontrol scheint mir an dieser Stelle z.B. aufgrund einer Prognose-Funktion viel intelligenter zu sein.

Habe ich das richtig verstanden, dass batcontrol mit dieser Erweiterung das Laden der Hausbatterie über diese Anbindung steuert, da es das viel intelligenter kann als evcc?

Ich danke Dir und viele Grüße

Alex

MaStr, 2024-10-26:
Hallo Alex,
Ja, evcc hat uns hier etwas „überholt“.
Jetzt ignorieren wir mal kurz die neue Funktion, dass evcc selbst nachlädt.

im Prinzip teile ich in meinem Kopf die Funktionen wie folgt auf:

**evcc**

- Überschüssigen PV Strom ins Auto bringen. Dabei die wallbox so steuern, dass z.b. Phasen geschaltet werden.
- Zu günstigen Tarifen das Auto laden (in Kombination mit Tibber oder ähnlichem).
- Die Batterie sperren, dass das Laden des eAutos nicht die Hausbatterie lädt.

**batcontrol**

- Die Hausbatterie nutzen um in den teuren Zeiten der dynamischen Tarife die Last über die Batterie zu regeln und Netzbezug zu meiden.
- Dafür wird in der Nacht zu günstigen Zeiten die Batterie gesperrt, sodass Ladestand für die teure Zeit (z.B. Uhr in der Früh) 
- Ist die Schwankung des Strompreises größer als der definierte Wert ( ich nehme 18 Cent an, da das gut Ladeverluste+Verschleiss abdeckt), dann lade zum günstigsten Zeitpunkt den Akku möglichst voll und verbrauche die Energie dann später.

Hierfür wird ein Standardlastprofil faktorisiert um deinen Jahresverbrauch, die erwartete Solarproduktion und sowie Batterie 
Stand gegeneinander verrechnet. Das wird dann gegen die aktuellen und nächsten Stunden des Strompreises verglichen und gehandelt.

Wir können aktuell evcc schon als Quelle für die Strompreise nehmen. Dieses issue hier ergänzt die Funktionen wie folgt:

- wenn evcc lädt, dann wird der Akku gesperrt, sobald der Ladestand < x% fällt.

Ohne diese Anbindung würde evcc sperren und batcontrol alle 2 Minuten entsperren.

MaStr, 2024-10-26:
Und wenn ich ganz verrückt denke, dann könnte man evcc auch als Wechselrichter-Schnittstelle implementieren und darüber dann laden und sperren steuern.

Ergänzend: im Gegensatz zu evcc bleibt hier nach stop der ursprüngliche Zeitplan des fronius erhalten, da muexx sich Mühe gegeben hat das zu sichern und wiederherzustellen.

WolfImBusch, 2024-10-26:
Super! Vielen Dank für Deine Erläuterung - jetzt hab ich' verstanden!!! Super!
Muss mir jetzt nur noch überlegen, wie ich batcontrol auf meiner Synology als Docker zum Laufen bekomme. Ich habe zwar auch HA/HACS - läuft aber ebenfalls als Docker und dann wird es mit der Installation schwieriger. Aber das bekomme ich schon irgendwie hin!! Viele Grüße aus Stuttgart!!! Alex

MaStr, 2024-11-15:
Heute live Test in meiner Anlage. Akku wird durch batcontrol gesperrt. Das funktioniert soweit.
Ich warte noch auf die Freigabe nach Beendigung des Ladevorgangs.

Was ich noch kontrollieren muss, ist wie sich evcc vs. mqtt-set API verhält.

MatzeBu, 2024-11-19:
> Und wenn ich ganz verrückt denke, dann könnte man evcc auch als Wechselrichter-Schnittstelle implementieren und darüber dann laden und sperren steuern.

Das wäre genial! 
Gibt es die Chance in dieser Richtung schon was zu Testen? Habe einen Sungrow Wechselrichter. Der Tickt mit EVCC soweit gut bzw auch mit HomeAssistant.
Eine sinnvolle Akkusteuerung fehlt mir bisher noch und das hier scheint ja nen super Ansatz zu sein.

MaStr, 2024-11-19:
@MatzeBu  Also ich bin gerade durch die EVCC API durchgegangen und es fehlen leider Schnittstellen um aktiv dann zu steuern.
Ich habe in EVCC auch die aktive Sperre der Batterie ausgestellt, da das bei mir batcontrol mit der Anbindung hier übernimmt.

Es gibt einen testtreiber, der hardcodierte Werte für die Batteriekapazität hat, aber bereits über MQTT einzelne Werte gesteuert werden könnte.
Wenn du den Testtreiber nimmst, kann man auf Basis theoretisch auch einen Inverter-Typ "MQTT" entwickeln, welcher dann von diversen Topics seine Werte Lesen kann und mit $magie die Informationen zum Schalten des Wechselrichters pusht. Aber das geht hier im Issue zu weit, glaube ich. :-)

MaStr, 2024-11-21:
Merged in [6e1cc0e](https://github.com/muexxl/batcontrol/commit/6e1cc0eacd3e18abdc92f88cf1c2f98419d187e5)

closed 2024-11-21 by MaStr