MaStr/batcontrol #12: Einspeisebegrenzung auf "0" setzen bei negativen Preisen

oe3cjb, 2024-08-28:
Nachdem du ja schon die variablen Energiepreise benutzt, wäre es eine feine Sache, wenn bei negativen Preisen die Einspeisebegrenzung auf 0W gesetzt wird. Sobald die Preise wieder ins positive gehen, dann wieder auf einen einstellbaren Wert.

MaStr, 2024-08-28:
Du meinst, dass bei negativen Börsenpreis nicht eingespeist werden soll, korrekt?

oe3cjb, 2024-08-28:
Ja genau - zur Zeit mache ich es manuell. Leider kann der Wert nicht über die API gesetzt werden, sondern nur über das Web-Interface. Mit der API habe ich schon Überschussladen programmiert - für http Programmierung in Python sind meine Kenntnisse zu schwach ;-)

MaStr, 2024-08-29:
Ok, verstanden.
Das werden wir aber nicht direkt mit dem Preis <0 realisieren können. Je nach Quelle kommt gleich der Preis+Gebühren an, also wäre bei mir z.B. 19 Cent die nulllinie.

Ich würde vorschlagen, dass das man das Preisziel, ab wann die Begrenzung, stattfindet, selbst einstellt. In meinem Fall 19 Cent.

Idealerweise hätte man eine Quelle für die Einspeisevergütung, sodass man das wirklich daran festmachen kann.

oe3cjb, 2024-08-29:
Soweit hatte ich gar nicht gedacht, da bei mir beim Einspeisen keine energieabhängigen Kosten dazukommen, sondern nur der monatliche (fixe) Preis für den Vertrag und ein prozentueller Abschlag. Der aber auch beim negativen Preis dann negativ ist – ich also weniger zahlen würde – daher ist bei mir 0 die Grenze.
Mit deinem Hintergrund macht dein Vorschlag natürlich definitiv Sinn, wenn sich jeder seine Einspeisegrenze selbst setzen kann.

MaStr, 2024-08-29:
Ja. Ich glaube man muss auch in der Diskussion bisschen differenzieren. Ich rede von meinem Verbrauchspreis. Das Skript selbst arbeitet nach dem Verbrauchspreis.

Beim Einspeisen ist im Deutschland noch ein fixer Wert für die kleinen Anlage zu haben. 
Die Einspeisevergütung kann ja tatsächlich variabel sein. Dann orientiert die sich mit aller Wahrscheinlichkeit letztlich auch am  Börsenpreis, vermute ich.

oe3cjb, 2024-08-29:
ja, genau - mein Einspeisetarif ist auch flexibel am Spot-Market-Preis angeknüpft und ist der gleiche Preis, wie mein Verbrauchspreis. Beim Bezug kommt dann noch die Netzgebühr dazu und beim Einspeisen holt sich mein EVU einen prozentuellen Abschlag von diesem Börsenpreis.

oe3cjb, 2024-08-29:
Hi Stephan,
ich selbst bin bei smartenergy.at (ein Tochterunternehmen der Energie Steiermark), die auch einen variablen Einspeisetarif anbieten. Der Abnahmepreis ist dabei basierend auf EPEX SPOT **AT** und smartenergy berechnet dafür einen Abschlag von 20%. Awattar bietet ein ähnliches Modell bei uns an.
Das mit dem Laden mit negativen Preis habe ich manuell auch schon genutzt - speziell mit meinem e-KFZ. Aufgrund der Netzgebühren von rund 6-7 ct/h macht das aber erst Sinn, wenn der Preis deutlich unter -7 ct/h sinkt, dann wird man dafür bezahlt, dass mein sein Auto auftankt :-D 
Ein Nachteil ist, dass die negativen Preise meistens nur in der Sommerzeit an den Wochenenden SA und SO ca. zw. 10:00 und 16:00 auftreten. Dann wenn der Speicher ohnehin immer gut voll und man sowieso autark ist. Natürlich könnte man dann bewusst laden und entladen/verkaufen, wenn es in der Nacht wieder teuer wird abzgl. des Verbrauchs, den man selber hat.
Smartenergy bietet auch eine API an, wobei die Preise die gleichen sind, wie bei Awattar - ich habe mal ein Beispiel von letzten Samstag beigefügt.
Ob sich das letztlich rechnet ist wahrscheinlich auch davon abhängig, wie die Lebensdauer des Speichers reduziert wird.
Mein Hauptthema ist eigentlich zu vermeiden, dass ich was zahle, wenn ich Einspeise - was ich momentan noch händisch erledige in dem ich das Soft Limit auf 0W setze.

[image]

muexxl, 2024-08-29:
Verstehe. 
Das heißt, eigentlich gibt es zwei Szenarien: 
1. Strompreis zwischen 0 und -7 ct/kWh, d.h. Negative Einspeisevergütung, aber positiver Preis bei Strombezug:
Einspeisung stoppen, Akku vollladen, falls nicht ohnehin voll

2. Strompreis kleiner -7 ct , also negative Einspeisevergütung, negativer Preis bei Strombezug:
Erzeugung stoppen, 
Verbrauch maximieren, 
Batterieladung halten

Und Szenario 1 ist hauptsächlich das relevante für dich, da es viel häufiger vorkommt. 
Stimmt das so?

oe3cjb, 2024-08-29:
zu 1) OK
zu 2) Batterieladung halten oder vollladen
genau 1) ist viel häufiger der Fall, dass der Preis so nach unten geht, dass 2) schlagend wird ist gaaaaanz selten

MaStr, 2024-11-18:
Ok, um das Einspeisen zu begrenzen muss man setzen:

POST `/config/batteries`
```
{"HYB_EM_MODE":1}
```

Dann ist im Menü hier die Felder frei:

[image]

 Mit

Ok, um das Einspeisen zu begrenzen muss man setzen:

POST `/config/batteries`
```{"HYB_EM_POWER":-100}```

Begrenze ich die Einspeisung auf 100W

Mit 

POST `/config/batteries`
```{"HYB_EM_POWER":100}```

Versucht er dann den Bezug bei 100W zu einzuschwingen.

Beim Wechsel alleine zwischen Automatisch & Manuell holt er eich bei Manuell den letzten Wert, den er kannte.

Also müsste eigentlich sowas hier gehen:

Mit 
`/config/batteries`
```
{"HYB_EM_MODE":0}
```
arbeitet er wieder "automatisch"

Also müsste man die Einspeise-Begrenzung mit 

POST `/config/batteries`
```
{"HYB_EM_MODE":1,
 "HYB_EM_POWER":0}
}
```
setzen.

MaStr, 2024-11-18:
Ich mag noch ergänzen, dass es vielleicht noch eine weitere Dimension gibt, die hier noch nicht so recht beleuchtet ist: Netzdienlichkeit.

Während man hier im Modell für das Einspeisen bei negativen Preisen "bestraft" wird, kann es auch Sinn machen die ganze Nummer auch netzdienlicher zu steuern.

- Ist im Forecast ein Sperrzeit der Einspeisung ersichtlich,
- Sperre/Drossele die Batterieladung
- Schalte die Ladung der Batterie so frei, dass am Ende des Sperrzeitraumes die Batterie bei X% .

Damit wäre die Anlage morgens netzdienlicher, weil bereits Solarstrom eingespeist wird. Gegen Mittag würde sie den Strom selbst in den Akku laden, wir müssten ihn auch nicht "verpuffen" lassen.

MaStr, 2024-11-29:
Tatsächlich brauche ich die config für einen anderen Fall, das hilft mir jetzt mehr als du glaubst 👍 
(Meine WP steht hinter dem Zähler, sodass der fronius nicht versucht mit der Batterie die WP zu versorgen (und ich 0,12s leer ist)). Aber zB die Umwältspumpe könnte man ja „füttern“.

MaStr, 2024-11-29:
@johannesghd  wir haben bei uns eine Grundfoss drin, die sollte in dem niedrigeren Bereich liegen.  Nur wenn die WP Taktet, dann ist halt Feierabend. Bei uns ist die Umwältspumpe in der Nibe integriert, sodass wir da keine Chance haben. 
Wir haben eine Kaskadenschaltung, sodass wir theoretisch die WP mit dem entsprechenden Tarif betreiben können. Aber das führt zu weit :-)
Ich weiß zumindest jetzt, dass ich darüber das nötige machen kann, wenn der dynamische Hausstrompreis unter den WP Tarif fällt.
Schreib mir bitte mal eine persönliche e-mail unter Matthias.strubel@aod-rpg.de 

aber grundsätzlich gilt, dass der Techniker Zugang Otto normalo nicht zur Verfügung steht, sodass es hier schwierig wird generell und pauschal in batcontrol zu lösen :-(

ich hätte als ersten Ansatz gewählt eine minim ladeleistung für die Batterie einzustellen und auf „ nicht ac laden“

MaStr, 2025-02-02:
Bin das Thema heute versuchsweise angegangen.
Das müsste über einen Umbau der Fronius Steuerung mit dem Customer Login möglich sein.

Die Batterie-Pläne, die wir aktuell ablegen sind immer "Ein Plan", also jetzt Sperren, Jetzt Entladen.

Wir müssten das so umbauen, dass die Fronius Klasse ein Steuerungsobjekt bekommt mit den Werten:

- Erlaubte Entladeleistung Batterie
- Erlaubte LadeLeistung Batterie
- Zwangsnachladen 

Damit setzt die Klasse dann die nötigen Schritte um.