MaStr/batcontrol #13: Telegram Bot

gjuecat, 2024-09-15:
Hallo,

zuerst mal vielen Dank für eure die Arbeit, ist wirklich schon ein tolles Tool.

Ich habe mit @muexxl eh schon mal zu Thema Telgram Bot geschrieben, aber aufgrund Zeitmangel konnte er das Thema bisher nie angehen. Nachdem nun beim Thema 'batcontrol' etwas mehr Bewegung reinkommt, hab ich mir gedacht, ich setze den Telegram Bot wieder mal auf die open issue Liste. Ich denke das wäre für viele (vor allem für den Einstieg und das Verständnis) schon sehr hilfreich, wenn man auf einfachem Weg (und auch unterwegs) den aktuellen Status sehen könnte.

lg
Jürgen

MaStr, 2024-09-15:
Hallo Jürgen,
kannst du etwas genauer erklären was du dir wünschst?
Wann soll eine Benachrichtigung passieren?
Was soll darin stehen?
Soll es interaktiv sein?

Liebe Grüße
Matthias

gjuecat, 2024-09-16:
Hallo Matthias,
wenn ich jetzt einen aktuell Status haben möchte, prüfe ich mit _systemctl status batcontrol.service_ die aktuellen Daten:
`● batcontrol.service - My Lovely Service
     Loaded: loaded (/lib/systemd/system/batcontrol.service; enabled; preset: enabled)
     Active: active (running) since Mon 2024-09-16 20:45:33 CEST; 10min ago
   Main PID: 127814 (python)
      Tasks: 1 (limit: 4596)
     Memory: 88.0M
        CPU: 765ms
     CGroup: /system.slice/batcontrol.service
             └─127814 python batcontrol.py

Sep 16 20:55:36 s36-debian-1 bash[127814]:   -939.72447156 -1376.72447156 -1753.72447156 -1756.54078594
Sep 16 20:55:36 s36-debian-1 bash[127814]:  -1875.54078594 -1983.54078594 -2080.54078594 -2157.72447156
Sep 16 20:55:36 s36-debian-1 bash[127814]:  -1611.72447156  -602.54078594   820.45921406  -111.72447156]
Sep 16 20:55:36 s36-debian-1 bash[127814]: 2024-09-16 20:55:36 DEBUG [BatCTRL] prices [0.15398472 0.14824968 0.13969656 0.12754668 0.132948   0.12921528
Sep 16 20:55:36 s36-debian-1 bash[127814]:  0.12551964 0.12173748 0.12153972 0.14714964 0.16041192 0.19757844
Sep 16 20:55:36 s36-debian-1 bash[127814]:  0.1908546  0.16111644 0.14768112 0.13247832 0.12251616 0.1167564
Sep 16 20:55:36 s36-debian-1 bash[127814]:  0.10814148 0.1182396  0.16175916 0.1540836  0.1540836  0.16530648]
Sep 16 20:55:36 s36-debian-1 bash[127814]: 2024-09-16 20:55:36 DEBUG [BatCTRL] Battery level (80.0) above discharge limit 50.0
Sep 16 20:55:36 s36-debian-1 bash[127814]: 2024-09-16 20:55:36 DEBUG [BatCTRL] Mode: Allow Discharging
Sep 16 20:55:36 s36-debian-1 bash[127814]: 2024-09-16 20:55:36 INFO [Fronius] Login successful
`

Ich verwende z.B. auch GEN24 Ladesteuerung - da gibt es die Möglichkeit, jede Datenänderung an den Wechselrichter als Telegram Message zu bekommen. So würde ich mir das hier auch vorstellen, über das Config-File konfiguraierbar, ob Nachrichten gesendet werden oder nicht.
Wenn sich dann z.B. das Ladelimit ändert, enladen gestoppt wird, nachladen erzwungen wird, ... hätte ich gerne eine Message auf Telegram. Ich stelle mir das gerade für debugging Zwecke oder allgemein zum Verstehen des Systems sehr hilfreich vor.

lg
Jürgen

MaStr, 2024-09-17:
Hallo Jürgen,
danke dir, das hilft schonmal.

Das „Verstehen“ ist übrigens auch der Grund warum ich die MQTT API eingebaut habe.

Ich schaue mir das bei Gelegenheit an.

Liebe Grüße Matthias

	Message ID: ***@***.***>