MaStr/batcontrol #17: Recent updates seems to break Forecast Solar without API key

OliJue, 2024-10-19:
An update within the last two weeks broke Forecast Solar without API key. I guess it was 0.2.1 and/or 0.2.3, because it changed the handling.

Log shows soon after start now:
`2024-10-19 21:31:50 WARNING [BatCtrl] Following Exception occurred when trying to get forecasts:`
`        unsupported operand type(s) for +: 'NoneType' and 'str'`

batcontrol_config.yaml contains the dummy entries:
`api: #fcsolarapi`

Do I need to update my batcontrol_config.yaml as well to work with the recent updates?

MaStr, 2024-10-20:
Pinning batcontrol to version-tag 0.2.3 & 0.2.5. , I'm receiving, the same error.
With latest main, not.

MaStr, 2024-10-20:
Ok, the circumvention of that bug is to comment that ``àpi: ``` line. The error is inside the configuration evaluation.

MaStr, 2024-10-20:
@muexxl   fixed that, please review and roll out a new version + HA-Adapter adjustment

commit 44c3aff5 (2024-10-20): Fixed #17
closed 2024-10-21 by muexxl