MaStr/batcontrol #18: Nachfrage, funktioniert always_allow_discharge_limit richtig?

MaStr, 2024-10-22:
```python3
    def is_discharge_allowed(self, net_consumption: np.ndarray, prices: dict):
        # always allow discharging when battery is >90% maxsoc
        discharge_limit = self.get_max_capacity() * self.always_allow_discharge_limit
        soc = self.get_SOC()
        if soc > discharge_limit:
            logger.debug(
                f'[BatCTRL] Battery level ({soc}) above discharge limit {discharge_limit}')
            return True
```

Ich nahm in meiner unendlichen Naivität an, dass get_SOC() den % Wert liefern soll, das scheint aber nicht korrekt zu sein, da das IF hier eigentlich den current_capacaty Wert braucht. Das wäre dann eigentlich "get_stored_energy()" was verlgichen wird, wenn ich mir die Berechnung vorher anschaue.

Ich hatte den get_SOC(), so wie oben im Testtreiber implementiert. Deswegen könnte das eine oder andere in der Umsellung auf die API Funktion gelitten haben.

Kannst du @muexxl  mir da bitte Feedback geben?

muexxl, 2024-10-22:
Ja nee, das macht so keinen Sinn hier wird SOC in der Range (0-100) mit dem discharge_limit in kWh verglichen. 
Muss man fixen.

closed 2024-10-22 by MaStr

muexxl, 2024-10-23:
Ok.. du warst einfach schneller :)

MaStr, 2024-10-23:
Hatte ich schon fertig. ;-)

commit 9fc1dcb5 (2024-10-22): Use correct value for is_discharge_allowed & is_discharge_allowed

Compare kWh with kWh and not SOC (%)

Fixed #18