MaStr/batcontrol #21: Docker image

MaStr, 2024-10-24:
Ich glaube ein docker image auf Basis von Alpine für x86 und arm64 fehlt uns auch noch.
Idealerweise gebaut via GitHub-Actions , wenn man den Tag der Version setzt.

muexxl, 2024-10-24:
Freut mich, dass du so Gas geben willst. 
Was hältst du davon, einfach mal einen Dev-Call zu machen in dem wir drüber reden, welche Features wir angehen wollen und wohin das Projekt gehen könnte?

MaStr, 2024-10-24:
Prinzipiell habe ich erstmal meine Gedanken der letzten Tage hier festgehalten.
Ich melde ich bei Dir Anfang nächste Woche, dann weiß Ich so grob wohin die Arbeitslast gehen wird.

MaStr, 2024-11-01:
Ich habe gestern Abend total Lust darauf gehabt und habe bei mir im Fork das soweit eingerichtet.

Ich habe ein Dockerfile und das GitHub action file erstellt. Ich würde die Commit bündeln und dann einen PR machen.

Das Action File triggert auf das Setzen von einem Tag mit dem Muster 0.0.0 .

Mir gefällt das tagging auf dem dockerhub noch nicht, weil ich gerne folgende Tags hätte:

- Latest
- 1.2.3
- 1.2
- 1

Entsprechend dem semantic versioning, kann der Verwender dann es handhaben wie sein watchtower das aktualisiert.

muexxl, 2024-11-01:
Gefällt mir. Hab aber mit git hub actions noch keine Erfahrungen. 
Es gibt ja noch das übergeordnete home assistant addon, welches auf den code in batcontrol verweist. 
Könnte es da Schwierigkeiten geben z.B., weil es dann zwei Dockerfiles in der Hierarchie gibt?

MaStr, 2024-11-01:
Ja, ich hatte das Dockerfile für den HomeAssitant Teil gesehen. Das habe ich frech kopiert und erweitert.
Nach meinem Wissensstand, landet bei der HA Variante dann das andere Dockerfile + github Actions des SourceRepository mit ins Image.

Ich habe noch ein paar Dinge, die erledigt werden sollten, bevor wir das Image dann auch generieren:

- [ ] Tags
- [ ] Entrypoint SH mit Prüfen auf die gesetzte Config mit verweis auf das Repository mit SampleConfig
- [ ] Laufen als non-root

muexxl, 2024-11-01:
Macht es dann nicht mehr Sinn ein getrenntes Repo für die non-HA Dockerversion zu haben ?

MaStr, 2024-11-01:
Das ist dann sehr kompliziert.
Du musst Zugriffsschlüssel zwischen den Repros teilen.
Hier wird ja jetzt automatisch der Bau in der Pipeline angestoßen, wenn du einen neuen Tag vergibst. Bei getrennten Repos muss dieses Repository dann einen Trigger auf das Bau-Repo schicken, damit dieses wiederum aktiv wird.

Ich könnte auch umgekehrt es formulieren: Macht es nicht mehr Sinn das Dockerfile für HA dazu zulegen oder anzupassen? Oder die Repository Struktur inhaltlich so zu ändern, dass der Source hier in "src" liegt und Templates und Dokumentation in "doc", "templates" etc. ?

MaStr, 2024-11-08:
I did not implement the image with an preset UserID.

MaStr, 2024-11-20:
Ich glaube ich habe die Lösungen für deine HomeAssistant dockerimage Fragen.

- Dein HA Image liegt im GitHub Repository, mein Image wird nach dockerhub gepusht.
- Um doppelte Änderungen zu vermeiden und das du den Code nicht verlinken musst, ist die Lösung die folgende:

Wir integrieren hier die Pipeline für ein Plain-Image ohne HA. Im HA repository verweist du im "from" nicht auf das BUILD_FROM Environment, sondern auf das plain docker-image. Als einzige Ergänzung bleibt das linken der options.json Datei auf die config.

Dann hat es die gleiche Basis. Was noch offen ist, auf welcher Basis wird das HA Image gebaut? Oder ist das ein plain alpine oder kommt der von der HA Pipeline was anderes rein?

MaStr, 2024-11-20:
Ok, typischerweise haben die Dinger wirklich ein eigenes Image dem supervisor s6 drin:

https://blog.michal.pawlik.dev/posts/smarthome/home-assistant-addons/

closed 2024-11-28 by MaStr

gjuecat, 2025-01-15:
Wie oft wird das Docker-Image aktualisiert? Ich denke da hat sich ja schon einiges getan, aber der Docker Container ist noch immer der von der (ersten) Version.

Kann ich das wo auslesen, auf welcher Version ich bin?

MaStr, 2025-01-15:
Beim Container Start wird die Version angedruckt, glaube ich.
Das Image mit dem Tag „main“ wird jeden Samstag neu gebaut.

Ein konkretes Release gab es bisher nicht nicht. Ich wollte noch das HomeAssistant Discovery Thema anfangen, daran hängt es derweil.