MaStr/batcontrol #22: get_max_capacity

muexxl, 2024-10-29:
Hey @MaStr 

ich habe gerade gesehen, dass in commit  d62045a3 get_max_soc() in get_max_capacity() umbenannt wurde. 

```
    def get_max_capacity(self):
        return self.max_soc
```
macht mMn keinen Sinn. 

Dadurch kam es auch zu. weil get_max_capacity eben den SOC zurückgibt
https://github.com/muexxl/batcontrol/issues/18

Und in baseclass.py
```
def get_max_capacity():
        raise RuntimeError("[Inverter Base Class] Function 'get_max_soc' not implemented")

```        
Und die aktuelle discharge_limit berechnung ist wieder falsch:
        discharge_limit = self.get_max_capacity() * self.always_allow_discharge_limit

Vorschlag: 
get_max_soc() wieder implementieren mit Ausgabe in der Range 0..100
get_max_capacity() brauchen wir mMn nicht. Da die max_capacity ja der capacity entspricht und es get_capacity schon gibt. 

alles, wo get_max_capacity() drin steckt muss jetzt wohl nochmal angeschaut worden. 

Den API teil überblicke ich nicht ganz. Deshalb will ich jetzt nicht einfach die umbenennung rückgängig machen. Hast dir ja bestimmt was dabei gedacht.

MaStr, 2024-10-29:
Oh fuck. Tut mir leid. Gut, dass du das 
```
    def get_max_capacity(self):
        return self.max_soc
``` 
gefunden hast.

Der Gedanke hinter get_max_capacity ist die Batterie-Kapazität abzubilden. Das vs. "stored energie", also wie viel habe ich davon in der Batterie noch übrig.
Get_max_soc macht so keinen Sinn, weil max_soc ist ja immer 100% :-D

Also bei einem 5kWh Akku wäre  ``get_max_capacity() = 5000 ``. 
Das wäre dann bei ``discharge_limit = self.get_max_capacity() * self.always_allow_discharge_limit `` korrekt so. Also z.B. nur bis 10% entladen, ergo 0.1 . Also 500Wh in dem Beispiel. Da ist dann auch  das hier korrekt

https://github.com/muexxl/batcontrol/blob/5168bff0451cc456822eab0551f7495e50046631/batcontrol.py#L401
```python
    def is_discharge_allowed(self, net_consumption: np.ndarray, prices: dict):
        # always allow discharging when battery is >90% maxsoc
        discharge_limit = self.get_max_capacity() * self.always_allow_discharge_limit
        stored_energy = self.get_stored_energy()
        if stored_energy > discharge_limit:
            logger.debug(
                f'[BatCTRL] Battery with {stored_energy} above discharge limit {discharge_limit}')
            return True
```

Der Bug ist aber auch nur im Froniuspy reingekommen. Im Testtreiber nicht.

MaStr, 2024-10-29:
ok, dann sollte man auch einen get_max_capacity() haben, der dann  ``(max_soc) /100 * nennkapazität`` abbildet. 

Da bin ich vielleicht etwas Naiv an die Sache mit der API herangegangen und etwas zu forsch refactored, oder auch kaputt gemacht.

MaStr, 2024-10-29:
Ich finde max_capacity da aber auch eher irreführend, weil warum ist dann max_capacity < capacity. 
Also ja, mit deiner Erklärung macht das Sinn, aber von aussen ist das eher unverständlich.

MaStr, 2024-10-29:
Frage: gilt nun:

always_discharge_limit (0.9)  gegen capacity (5kWh)  =  4,5 kWh

oder

always_discharge_limit * capacity * max_soc (0.95) = 4,275 kWh

?

muexxl, 2024-10-29:
wozu braucht man denn  get_max_capacity() = (max_soc) /100 * nennkapazität ?
die drei reichen doch eigentlich:
get_free_capacity() - (max_soc - current_soc ) /100 * nennkapazität
get_stored_energy() -(current_soc - min_soc ) /100 * nennkapazität
get_usable_capacity() - (max_soc - min_soc ) /100 * nennkapazität
oder nicht?

always_discharge_limit (0.9) gegen capacity (5kWh) = 4,5 kWh
oder
always_discharge_limit * capacity * max_soc (0.95) = 4,275 kWh

meine Idee war, dass sich always_discharge_limit als SOC Wert versteht. 
Also wenn always_discharge_limit > SOC, dann auf jeden fall entladen. 
Wegen der unterschiedlichen Skalen  wird das dann zu always_discharge_limit > SOC/100 

Sprich, wenn die Batterie eh fast voll ist, dann kann ich auch entladen. Egal was die Preise etc. sagen. 

Falls man max_soc < always_discharge_limit gesetzt hat, dann wäre das Limit somit wirkungslos.

MaStr, 2024-10-29:
> meine Idee war, dass sich always_discharge_limit als SOC Wert versteht.
> ...
> Falls man max_soc < always_discharge_limit gesetzt hat, dann wäre das Limit somit wirkungslos.

Ok, also sind alle Angaben in der Config quasi % gegen die maximale Kapazität.  Das war mir jetzt noch wichtig zu verstehen.

Dann muss man hier

https://github.com/muexxl/batcontrol/blob/5168bff0451cc456822eab0551f7495e50046631/batcontrol.py#L256
https://github.com/muexxl/batcontrol/blob/5168bff0451cc456822eab0551f7495e50046631/batcontrol.py#L326
https://github.com/muexxl/batcontrol/blob/5168bff0451cc456822eab0551f7495e50046631/batcontrol.py#L403

Nochmal ran, weil das jetzt gegen max_capacity geht und damit nicht das von dir beschriebene Verhalten hätte.

So gesehen brauch man es im Code direkt nicht unbedingt. Was aber aktuell batcontrol fehlt, ist einen Wert der die gesamte installierte Kapazität und verfügbare Kapazität (nicht Ladung, also max_capacity) widerspiegelt. Den Wert braucht man vielleicht nicht direkt in der Berechnung, aber wenn wir die über die API mit rausschreiben, kann man das innerhalb von einem Grafana-Dashboard auch sichtbar machen.
Wenn wir die Werte "im Bauch" sichtbar machen, hilft es hoffentlich mehr nachzuvollziehen was hier passiert.

Vom Gedanken her habe ich die Methoden wie folgt strukturiert:

1. batcontrol - get_usable_capacity - Alle Inverter addiert (gut, wir haben erstmal nur einen)
2. inverter/get_usable_capacity - von der Inverter Instanz.

Das ist gedanklich die Vorbereitung ggf. zwei oder mehr Inverter zu steuern. Da ich eh ran musste, habe ich es dann gleich so abstrahiert.

MaStr, 2024-10-30:
Das ist auf Main bereits korrigiert und halt noch nicht als Release geflagged worden, weil wir gerade noch am Besprechen über die genaue Implementierung waren, da noch Änderungen nötig sind.

Wegen dem stable branch, könne wir machen, wäre aber auch redundant.
Wir haben derzeit Entwicklung auf dem main Branch und du kannst die Tags der jeweiligen Version auschecken.

@muexxl bump wegen hotfix

MaStr, 2024-10-30:
Ok, sorry, ich dachte ich hätte den eine commit rübergezogen.

MaStr, 2024-10-30:
Ich hab die eine Änderung jetzt per cherry-pick rübergeholt.

PhilippBLG, 2024-11-01:
Könntet ihr den commit auch auf das ha_addon pushen? Oder gibt es eine weitere Möglichkeit die aktuelle Version auf Home Assistant zum laufen zu bekommen?

MaStr, 2024-11-01:
Ich denke, dass @muexxl möglichst zeitnah eine neue Version bereitstellt.
Für den HomeAssistant geht der Fix wohl nicht einfach so mit der Hand rein.

MaStr, 2024-11-14:
> Dann muss man hier
>
> https://github.com/muexxl/batcontrol/blob/5168bff0451cc456822eab0551f7495e50046631/batcontrol.py#L256 [...]

Diese Stellen würde ich umstellen von max_capacity auf capacity() , was dann den Wert abzüglich MAX_SOC entspricht.
Dann könnten wir hier auch zu machen. 

OK @muexxl  ?

MaStr, 2024-11-20:
Also ich bin der Meinung get_max_capacity() muss bleiben.

Capacity ist in fronius.py das hier
https://github.com/muexxl/batcontrol/blob/b64434f59295fd1c8489bafead10b780b9bb5130/inverter/fronius.py#L346-L348

Also die Designed capacity.

Während:
https://github.com/muexxl/batcontrol/blob/b64434f59295fd1c8489bafead10b780b9bb5130/inverter/fronius.py#L105C1-L106C35

Das hier MAX_SOC berücksichtigt.

Ich mache dann hier zu.

closed 2024-11-20 by MaStr