MaStr/batcontrol #25: Logfile Limit via Python handler

MaStr, 2024-10-31:
Instead of an own class for log handling, make use of the default RotatingFileHandler from python.

muexxl, 2024-11-03:
Ich habe mir den RotatingFilehandler angeschaut und dagegen entschieden. 

## Begründung
Das Problem ist ja, dass der Speicher irgendwann von den Logs aufgezehrt wird. Bei Rotating logs habe ich den gleichen Speicherverbrauch, nur eben auf mehrere Dateien verteilt. Für das Löschen der einzelnen Files bräuchte man dann wieder ein entsprechendes Skript.

MaStr, 2024-11-03:
Wieso? Du kannst doch ein Generationen Limit eintragen.
https://docs.python.org/3/library/logging.handlers.html#logging.handlers.RotatingFileHandler

BackupCount=1 , ist eine zweite Datei , im Wechsel A/B
Maxbytes=100

Oder du stellst auf 50kb und backupcount=1, dann hast Du die 100kb.

Ich verstehe dein Ansinnen, aber es macht für mich wenig Sinn hier das Rad neuzuerfinden.

muexxl, 2024-11-03:
Ok. Vielleicht hatte ich mir das einfach nicht genau genug angeschaut. 
Ich sehe jezt aber keinen direkten Mehrwert darin, das nochmal zu ändern.
Aber hey.. feel free to implement.