MaStr/batcontrol #32: crash when pulling powerconfig

muexxl, 2024-11-03:
Linked to PR29 https://github.com/muexxl/batcontrol/pull/29

Der Wechselrichter antworte mit "Bad Request" offenbar stimmt etwas mit der API Abfrage nicht. 

```
Traceback (most recent call last):
  File "/batcontrol/./batcontrol.py", line 644, in <module>
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/./batcontrol.py", line 98, in __init__
    self.inverter = inverter.Inverter(config['inverter'])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/inverter.py", line 13, in __new__
    return FroniusWR(config['address'], config['user'], config['password'], config['max_grid_charge_rate'], config['max_pv_charge_rate'])
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 49, in __init__
    self.previous_backup_power_config = self.get_powerunit_config()
                                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 105, in get_powerunit_config
    response = self.send_request(path, auth=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 347, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Request failed with 400-Bad Request. 
	url:http://10.0.0.66/config/powerunit, 
	params:None 
	headers {'Authorization': 'Digest username="customer", realm="Webinterface area", nonce="6727cc90:aa63ec3c073772e8b24e91a8ddf1d344", uri="/config/powerunit", algorithm="MD5", qop=auth, nc=00000001, cnonce="NaN", response="51520aebad0d429ff564d27d42316186"'} 
	nonce 6727cc90:aa63ec3c073772e8b24e91a8ddf1d344 
	payload 
2024-11-03 20:18:40 INFO [Inverter] Restoring previous battery configuration: {"BAT_M0_SOC_MAX": 100, "BAT_M0_SOC_MIN": 8, "BAT_M0_SOC_MODE": "manual", "HYB_BM_CHARGEFROMAC": false, "HYB_EM_MODE": 1, "HYB_EM_POWER": 0, "HYB_EVU_CHARGEFROMGRID": false} 
2024-11-03 20:18:40 ERROR [Inverter] could not restore timeofuse config
2024-11-03 20:18:40 INFO [Inverter] Logout successful
```

muexxl, 2024-11-03:
in GET /status/version gibt es swrevisions.GEN24 oder und ein Array apiversions ..

da es -bisher- nur der eine call ist könnte man ganz pragmatisch auch config/powerunit callen und falls es einen 400er response code gibt die legacy uri config/setup/powerunit probieren.

hab jetzt auch den WR geupdated und es funktioniert wieder. 
Wir sollten das aber auf jeden Fall fixen, damit auch die 1.2.xer Versionen noch unterstützt werden.
Das ist halt der sh** wenn man undokumentierte APIs reengineert ..

MaStr, 2024-11-03:
Die Solar-API scheint zumindest dokumentiert. https://www.fronius.com/~/downloads/Solar%20Energy/Operating%20Instructions/42,0410,2012.pdf

Aber es scheint wir nutzen eine andere. 

Wollte es nur mal teilen.

muexxl, 2024-11-04:
"Wir nutzen eine andere" -> kann man so sagen :D. 

Ich habe mir halt die Kommunikation vom WebUI mit dem Backend angeschaut und dann re-engineered. Die SolarAPI von Fronius kann man nur nutzen um Parameter auszulesen, aber nicht um Werte zu setzen. 

Später habe ich dann erfahren, dass es wohl auch eine elegantere Möglichkeit via Modbus TCP gibt.

MaStr, 2024-11-04:
Auch wenn es nicht direkt hier reingehört: Wir sollten uns angewöhnen solche grundlegenden Changes mit einem Featuer-Flag erst zu versehen, dass die Applikation wenigstens lauffähig bleibt.

MaStr, 2024-11-04:
Schwierig. Du kannst mit der gefangenen Exception unter Umständen auch nichts anfangen.
Gerade wenn du bei der Initialisierung bist, dann ist Ende, weil du keine Daten bekommst.

"Unterwegs" kann man das machen. Da könnte man zurückfallen in die Verarbeitungsschleife und nach 2 Minuten es erneut versuchen (z.B. WR Restart). Da wäre sogar das Beenden vom batcontrol kontraproduktiv, weil das Einspielen der alten Konfiguration fehlschlagen würde.

MaStr, 2024-11-05:
@johannesghd  Kannst du dir die version auf dem Branch issue-32 bitte anschauen?  Damit sollte es ja tun.

MaStr, 2024-11-13:
Merged into Release-0.2.11 branch for testing

commit 12f967d1 (2024-11-14): Fallback to old powerunit config url

Fixed #32
commit ac994c3a (2024-11-14): Enhance backup power configuration handling in FroniusWR class

- Fronius.py : Add an additional message in case of fallback URL
- Fallback to old powerunit config url

Fixed #32
closed 2024-11-14 by muexxl