MaStr/batcontrol #41: Problem bei Batterieladen per Netzbezug bei günstigen dynamischen Preis

normanSp1606, 2024-11-06:
Hallo zusammen,
ich habe eine Frage zur bzw. Problem bei der Funktionsweise von batcontrol.
Bei niedrigem Akkustand und geringer prognostizierter PV-Energie ist meine Annahme, dass der Speicher mit Netzstrom geladen wird, wenn die Strompreise niedrig sind.

Meine Vermutung aus den letzten Tage ist:
- bisher erfolgt garkein Laden des Speichers aus Netz
- im besten Fall: das Entladen wird vermieden und Bedarf durch Netzbezug gedeckt
- ein Verhalten ist mir noch aufgefallen: sobald etwas PV-Strom erzeugt wird, wird damit nur der Speicher aufgeladen, der aktuelle Verbrauch mehr oder weniger vollständig per Netzbezug gedeckt.

Hier sind zwei Screenshots zum Lade/Entlade-Verhalten des Speichers und der dazugehörigen Strompreise. Über Nacht ist der Speicher komplett entladen worden. Bei korrekter Funktionsweise hätte er ja vor der morgendlichen Preisspitze mit Netzstrom aufgeladen werden müssen, um zumindest bis 10 Uhr zu kommen, oder sehe ich das falsch?

Ist in der Config etwas falsch, oder muss ich im Fronius etwas freigeben, um Netzladen zuzulassen?

Zu meinem Setup:
- home-assistant
- batcontrol addon
- evcc läuft auch noch parallel (kann das Wechselwirkungen erzeugen? evcc hat aber keine laufenden Ladevorgänge)

Danke schonmal vorab :) 

Meine Konfiguration (über home assistant addon UI):

```
min_price_difference: 0.05
always_allow_discharge_limit: 0.99
max_charging_from_grid_limit: 0.8

type: fronius_gen24
address: 192.168.188.80
user: customer
password: **************
max_pv_charge_rate: 5000
max_grid_charge_rate: 5000
max_charge_rate: 5000
max_grid_power: 5000
type: tibber
apikey: **********
vat: 0.2
fees: 0.015
markup: 0.03

- name: HA13
  lat: 51.****
  lon: 11.*****
  declination: 25
  azimuth: 0
  kWp: 5.4
  api: https://api.forecast.solar/estimate/51.99/11.99
![IMG_2D4EE8C13C2C-1](https://github.com/user-attachments/assets/4e924888-964b-4d6e-97f7-0c22946ef1f8)
![IMG_6CBEDB0E4B57-1](https://github.com/user-attachments/assets/5a0e3c1b-268b-4963-ad4c-cdee2071d12b)
/20/0/5.4

annual_consumption: 4500
load_profile: load_profile.csv

```
[image]
[image]

MaStr, 2024-11-06:
EVCC kann Wechselwirkungen haben. Gerade in den letzten Versionen wurde auch aktives Batterie-Management ergänzt.

Wie hast du EVCC konfiguriert? 
Welche API nutzt du im EVCC um an die Daten zu kommen?
Hast du das Logfile von batcontrol aus der Nacht?

thomeyer1, 2024-11-06:
Hi, habe gerade einen issue eingefügt, bei mir selbes Verhalten, evcc aber auf einem andere Pi, habe mal eine Kleinigkeit geändert, da bei mir das AddOn auch immer abgeflogen ist, genau wenn das Laden starten sollte. sieht man leider nur, wenn man das Container Log anschaut. charge_rate ist bei mir ein floating Wert, wie bei Dir: 799.5465557556231 (bei mir ein anderer beliebiger).
Habe in meinem Issue einen Ansatz dokumentiert, wie ich es jetzt mache und siehe da, die Batterie wurde erstmals aus dem Grid geladen.

Gruß Thomas

Traceback (most recent call last):
  File "/batcontrol/./batcontrol.py", line 647, in <module>
    bc.run()
  File "/batcontrol/./batcontrol.py", line 309, in run
    self.set_wr_parameters(net_consumption, price_dict)
  File "/batcontrol/./batcontrol.py", line 337, in set_wr_parameters
    self.force_charge(charge_rate)
  File "/batcontrol/./batcontrol.py", line 507, in force_charge
    self._set_charge_rate(charge_rate)
TypeError: 'int' object is not callable

thomeyer1, 2024-11-06:
Container Log: docker logs addon_a31385ec_batcontrol | more

thomeyer1, 2024-11-06:
ja, dachte ich auch, habe bei mir im Container beide Stellen geändert und beobachte das mal. Screenshot kommt. Das AddOn überschreibt meine Einstellungen ja 
[image]

thomeyer1, 2024-11-06:
Laden hat er auf "On" gestellt, klassisch immer auf "Off"

thomeyer1, 2024-11-06:
neuste verfügbare, gestern update gemacht
in den Docker Container kommst Du zum ändern mit "docker exec -it addon_a31385ec_batcontrol sh"
dann habe ich das batcontrol.py File geändert

thomeyer1, 2024-11-06:
SolarWeb Sicht - lädt gerade wieder
[image]

thomeyer1, 2024-11-06:
aber das läuft doch als HomeAssistant AddOn? dann ist es Docker

thomeyer1, 2024-11-06:
gehe mal is Terminal und "docker ps"

thomeyer1, 2024-11-06:
ok

MaStr, 2024-11-06:
Dann können wir ja zu machen, weil das sch gefixed ist.

thomeyer1, 2024-11-06:
lasst mich mal wissen, ob #34 alleine helfen würde, bitte mal testen, allerdings kommen wir da ja von einer ganz anderen Stelle

MaStr, 2024-11-06:
Ich mache hier nochmal auf, bis ihr endgültig das Problem gefunden habt.

normanSp1606, 2024-11-06:
Guten Abend zusammen, 
und danke für eure Hilfe.

> > * ein Verhalten ist mir noch aufgefallen: sobald etwas PV-Strom erzeugt wird, wird damit nur der Speicher aufgeladen, der aktuelle Verbrauch mehr oder weniger vollständig per Netzbezug gedeckt.
> 
> Hast du hierzu einen Screenshot aus der Wechselrichter-IP/#/battery-control-schedule ?
> 
> Unter Umständen ist die Einstellung "Batterieladung aus anderen Quellen" nicht richtig konfiguriert.

Das war tatsächlich ein Problem. Die Einstellung hatte ich schonmal gemacht, heute aber festgestellt, dass die Firmware des WR diese nicht gespeichert hat bzw. auch heute nicht dauerhaft speichern wollte. Nach FW-Update bleibt die Einstellung nun auch dauerhaft drin.
Home Assistant auch nochmal geupdatet.
Bisher sind in den Logs keine Exceptions zu sehen. Kann aber auch daran liegen, dass gerade kein Nachladen über Netz sinnvoll erscheint. Der Vollständigkeit halber, anbei das aktuellste Log. 
[batcontrol_log.txt](https://github.com/user-attachments/files/17652353/batcontrol_log.txt)

Ich werd den nächsten Tag nochmal beobachten bzw. auf den Fix in PR #34 warten.

MaStr, 2024-11-06:
Das logfile sieht aber nicht gut aus.
Zumindest laut log ist die Batterie leer. Seit kurz vor acht kommt er zu dem Schluss, dass er keine zusätzliche Energie vorhalten muss. 
Er sperrt einfach die Batterie vor entladen.
Gleichzeitig berücksichtigt der nur die aktuelle Stunde, weil ab 21:00 Uhr der Preis wieder günstiger wird.

Schlechtes Beispiel vielleicht…? 

Generell läufst du hier in die Falle, die wir einem anderen Ticket schon durchdiskutiert haben bezüglich der Lademengen Diskussion.

thomeyer1, 2024-11-07:
Guten Morgen,
bei mir lädt er nur ein bisschen und gestern zum Teil bei Preisspitzen

Logausschnitt
2024-11-06 15:11:41 DEBUG [FC Cons] predicting consumption {0: 341.813414870701, 1: 338.34891743634853, 2: 370.20313346766676, 3: 433.90570054468384, 4: 335.69510944304756, 5: 339.413203044421, 6: 305.5866134345802, 7: 258.02736888332174, 8: 288.4664931440413, 9: 244.49112014884665, 10: 212.18041203763255, 11: 236.11772378411953, 12: 236.08699610128372, 13: 369.63566424400517, 14: 329.8575897871232, 15: 326.05918272655765, 16: 318.2241995323487, 17: 327.8346770166804, 18: 345.10881219096706, 19: 349.31066637990733, 20: 337.44065310859133, 21: 449.94174003422916, 22: 431.4012223906464, 23: 429.96383934717215, 24: 366.52213189636865, 25: 385.9124414911546}
2024-11-06 15:11:41 DEBUG [BatCTRL] Production FCST [ 512.  110.    0.    0.    0.    0.    0.    0.    0.    0.    0.    0.
    0.    0.    0.    0.  269.  901. 1193. 1329. 1322. 1196. 1013.  785.
  523.  107.]
2024-11-06 15:11:41 DEBUG [BatCTRL] Consumption FCST [341.81341487 338.34891744 370.20313347 433.90570054 335.69510944
 339.41320304 305.58661343 258.02736888 288.46649314 244.49112015
 212.18041204 236.11772378 236.0869961  369.63566424 329.85758979
 326.05918273 318.22419953 327.83467702 345.10881219 349.31066638
 337.44065311 449.94174003 431.40122239 429.96383935 366.5221319
 385.91244149]
2024-11-06 15:11:41 DEBUG [BatCTRL] Net Consumption FCST [-170.18658513  228.34891744  370.20313347  433.90570054  335.69510944
  339.41320304  305.58661343  258.02736888  288.46649314  244.49112015
  212.18041204  236.11772378  236.0869961   369.63566424  329.85758979
  326.05918273   49.22419953 -573.16532298 -847.89118781 -979.68933362
 -984.55934689 -746.05825997 -581.59877761 -355.03616065 -156.4778681
  278.91244149]
2024-11-06 15:11:41 DEBUG [BatCTRL] prices [0.4914 0.7938 1.1459 1.128  0.7054 0.4167 0.3288 0.312  0.3029 0.3003
 0.2925 0.2886 0.2888 0.2922 0.309  0.3573 0.4312 0.3984 0.3519 0.3471
 0.3304 0.3307 0.3281 0.3365 0.3954 0.4807]
2024-11-06 15:11:41 DEBUG [BatCTRL] Evaluating next 5 hours until 19:59
2024-11-06 15:11:41 DEBUG [BatCTRL] Reserved Energy: 1229.2 Wh. Available in Battery: 890.9Wh
2024-11-06 15:11:41 DEBUG [BatCTRL] Discharging is NOT allowed
2024-11-06 15:11:41 DEBUG [BatCTRL] Charging allowed: True
2024-11-06 15:11:41 DEBUG [BatCTRL] Additional Energy required: 0.0 Wh
2024-11-06 15:11:41 DEBUG [BatCTRL] Mode: Avoid Discharging
2024-11-06 15:13:41 DEBUG [FC Cons] predicting consumption {0: 341.813414870701, 1: 338.34891743634853, 2: 370.20313346766676, 3: 433.90570054468384, 4: 335.69510944304756, 5: 339.413203044421, 6: 305.5866134345802, 7: 258.02736888332174, 8: 288.4664931440413, 9: 244.49112014884665, 10: 212.18041203763255, 11: 236.11772378411953, 12: 236.08699610128372, 13: 369.63566424400517, 14: 329.8575897871232, 15: 326.05918272655765, 16: 318.2241995323487, 17: 327.8346770166804, 18: 345.10881219096706, 19: 349.31066637990733, 20: 337.44065310859133, 21: 449.94174003422916, 22: 431.4012223906464, 23: 429.96383934717215, 24: 366.52213189636865, 25: 385.9124414911546}
2024-11-06 15:13:41 DEBUG [BatCTRL] Production FCST [ 512.  110.    0.    0.    0.    0.    0.    0.    0.    0.    0.    0.
    0.    0.    0.    0.  269.  901. 1193. 1329. 1322. 1196. 1013.  785.
  523.  107.]
2024-11-06 15:13:41 DEBUG [BatCTRL] Consumption FCST [341.81341487 338.34891744 370.20313347 433.90570054 335.69510944
 339.41320304 305.58661343 258.02736888 288.46649314 244.49112015
 212.18041204 236.11772378 236.0869961  369.63566424 329.85758979
 326.05918273 318.22419953 327.83467702 345.10881219 349.31066638
 337.44065311 449.94174003 431.40122239 429.96383935 366.5221319
 385.91244149]
2024-11-06 15:13:41 DEBUG [BatCTRL] Net Consumption FCST [-170.18658513  228.34891744  370.20313347  433.90570054  335.69510944
  339.41320304  305.58661343  258.02736888  288.46649314  244.49112015
  212.18041204  236.11772378  236.0869961   369.63566424  329.85758979
  326.05918273   49.22419953 -573.16532298 -847.89118781 -979.68933362
 -984.55934689 -746.05825997 -581.59877761 -355.03616065 -156.4778681
  278.91244149]
2024-11-06 15:13:41 DEBUG [BatCTRL] prices [0.4914 0.7938 1.1459 1.128  0.7054 0.4167 0.3288 0.312  0.3029 0.3003
 0.2925 0.2886 0.2888 0.2922 0.309  0.3573 0.4312 0.3984 0.3519 0.3471
--More-- (10% of 1107904 bytes)

normanSp1606, 2024-11-07:
Hallo,
Es scheint so, dass es (mindestens beser) funktioniert als gestern.
Zwischen 2 und 5 Uhr wurde ca. 1kWh aus dem Netz in den Speicher geladen. Hat nicht ganz gereicht um die Lastspitze heute morgen komplett aus dem Speicher abzudecken. Kann aber auch am Lastprofil liegeen oder wie eng die Optimierung arbeitet und wie/ob Puffer eingeplant werden.

[image]

Leider sind die Logs von heute nicht da, scheinbar ist der Container/App abgeraucht und neu gestartet.

Die Logs von heute morgen:
```
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-11-07 08:53:27 INFO [FCConsumption] loading module
2024-11-07 08:53:27 INFO [FCSolar] loading module
2024-11-07 08:53:27 INFO [LogFileLimiter] loading module
2024-11-07 08:53:27 INFO [Main] Starting Batcontrol
2024-11-07 08:53:27 INFO [Batcontrol] host system time zone was not set. Setting to Europe/Berlin
2024-11-07 09:53:27 INFO [Inverter] loading module
2024-11-07 09:53:27 INFO [Inverter] Login successful
2024-11-07 09:53:28 INFO [FC Cons] The annual consumption of the applied load profile is 3225.2876977966316 kWh
2024-11-07 09:53:28 INFO [FC Cons] the hourly values from the load profile are scaled with a factor of 1.40 to match the annual consumption of 4500.0 >
2024-11-07 09:53:28 INFO [FCSolar] Requesting Information for PV Installation HA13
2024-11-07 09:53:29 DEBUG [FC Cons] predicting consumption {0: 345.10881219096706, 1: 349.31066637990733, 2: 337.44065310859133, 3: 449.94174003422916>
2024-11-07 09:53:29 DEBUG [BatCTRL] Production FCST [ 561.  876. 1097. 1207. 1190. 1056.  801.  212.    0.    0.    0.    0.
    0.    0.    0.]
2024-11-07 09:53:29 DEBUG [BatCTRL] Consumption FCST [345.10881219 349.31066638 337.44065311 449.94174003 431.40122239
 429.96383935 366.5221319  385.91244149 377.04416591 332.00330903
 410.72932605 530.62027751 478.66383631 321.70422592 234.85612362]
2024-11-07 09:53:29 DEBUG [BatCTRL] Net Consumption FCST [-215.89118781 -526.68933362 -759.55934689 -757.05825997 -758.59877761
 -626.03616065 -434.4778681   173.91244149  377.04416591  332.00330903
  410.72932605  530.62027751  478.66383631  321.70422592  234.85612362]
2024-11-07 09:53:29 DEBUG [BatCTRL] prices [0.3535 0.3487 0.3319 0.3322 0.3297 0.338  0.3969 0.4823 0.6573 0.5247
 0.3799 0.3255 0.3071 0.3045 0.2903]
2024-11-07 09:53:29 DEBUG [BatCTRL] Evaluating next 14 hours until 22:59
2024-11-07 09:53:29 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: -11.1Wh
2024-11-07 09:53:29 DEBUG [BatCTRL] Discharging is NOT allowed
2024-11-07 09:53:29 DEBUG [BatCTRL] Charging allowed: True
2024-11-07 09:53:29 DEBUG [BatCTRL] Additional Energy required: 11.1 Wh
2024-11-07 09:53:29 DEBUG [BatCTRL] Mode: grid charging. Charge rate : 94.79142857142824 W
2024-11-07 09:55:30 DEBUG [FC Cons] predicting consumption {0: 345.10881219096706, 1: 349.31066637990733, 2: 337.44065310859133, 3: 449.94174003422916>
2024-11-07 09:55:30 DEBUG [BatCTRL] Production FCST [ 561.  876. 1097. 1207. 1190. 1056.  801.  212.    0.    0.    0.    0.
    0.    0.    0.]
2024-11-07 09:55:30 DEBUG [BatCTRL] Consumption FCST [345.10881219 349.31066638 337.44065311 449.94174003 431.40122239
 429.96383935 366.5221319  385.91244149 377.04416591 332.00330903
 410.72932605 530.62027751 478.66383631 321.70422592 234.85612362]
2024-11-07 09:55:30 DEBUG [BatCTRL] Net Consumption FCST [-215.89118781 -526.68933362 -759.55934689 -757.05825997 -758.59877761
 -626.03616065 -434.4778681   173.91244149  377.04416591  332.00330903
  410.72932605  530.62027751  478.66383631  321.70422592  234.85612362]
2024-11-07 09:55:30 DEBUG [BatCTRL] prices [0.3535 0.3487 0.3319 0.3322 0.3297 0.338  0.3969 0.4823 0.6573 0.5247
 0.3799 0.3255 0.3071 0.3045 0.2903]
2024-11-07 09:55:30 DEBUG [BatCTRL] Evaluating next 14 hours until 22:59
2024-11-07 09:55:30 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 0.0Wh
2024-11-07 09:55:31 DEBUG [BatCTRL] Discharging is NOT allowed
2024-11-07 09:55:31 DEBUG [BatCTRL] Charging allowed: True
2024-11-07 09:55:31 DEBUG [BatCTRL] Additional Energy required: 0.0 Wh
2024-11-07 09:55:31 DEBUG [BatCTRL] Mode: Avoid Discharging
2024-11-07 09:57:31 DEBUG [FC Cons] predicting consumption {0: 345.10881219096706, 1: 349.31066637990733, 2: 337.44065310859133, 3: 449.94174003422916>
2024-11-07 09:57:31 DEBUG [BatCTRL] Production FCST [ 561.  876. 1097. 1207. 1190. 1056.  801.  212.    0.    0.    0.    0.
    0.    0.    0.]
2024-11-07 09:57:31 DEBUG [BatCTRL] Consumption FCST [345.10881219 349.31066638 337.44065311 449.94174003 431.40122239
 429.96383935 366.5221319  385.91244149 377.04416591 332.00330903
 410.72932605 530.62027751 478.66383631 321.70422592 234.85612362]
2024-11-07 09:57:31 DEBUG [BatCTRL] Net Consumption FCST [-215.89118781 -526.68933362 -759.55934689 -757.05825997 -758.59877761
 -626.03616065 -434.4778681   173.91244149  377.04416591  332.00330903
[... 272 lines cut ...]
 530.62027751 478.66383631 321.70422592 234.85612362]
2024-11-07 10:31:45 DEBUG [BatCTRL] Net Consumption FCST [-526.68933362 -759.55934689 -757.05825997 -758.59877761 -626.03616065
[... cut ...]
 -434.4778681   173.91244149  377.04416591  332.00330903  410.72932605
  530.62027751  478.66383631  321.70422592  234.85612362]
2024-11-07 10:33:46 DEBUG [BatCTRL] prices [0.3487 0.3319 0.3322 0.3297 0.338  0.3969 0.4823 0.6573 0.5247 0.3799
 0.3255 0.3071 0.3045 0.2903]
2024-11-07 10:33:46 DEBUG [BatCTRL] Evaluating next 13 hours until 22:59
2024-11-07 10:33:46 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 22.1Wh
2024-11-07 10:33:46 DEBUG [BatCTRL] Mode: Allow Discharging
2024-11-07 10:33:46 INFO [Inverter] Login successful
```

MaStr, 2024-11-07:
Die Optimierung arbeitet komplett gegen dein Lastprofil. Ist das Lastprofil zu niedrig, dann lädt auch nicht genug nach.

gjuecat, 2024-11-07:
Hallo,
ich hätte mir gestern aber auch heute ein "Vorladen" der Batterie aufgrund der hohen Strompreise am Abend erwartet. Ist aber bisher nicht erfolgt.
[batcontrol.log](https://github.com/user-attachments/files/17660435/batcontrol.log)

batcontrol_config.yaml
```
timezone: Europe/Berlin #your time zone. not optional.
loglevel: debug
max_logfile_size: 100 #kB
battery_control:
  min_price_difference: 0.07 # minimum price difference in Euro to justify charging your battery
  always_allow_discharge_limit: 0.50 # 0.00 to 1.00 above this SOC limit using energy from the battery is always allowed
  max_charging_from_grid_limit: 0.90 # 0.00 to 1.00 charging from the grid is only allowed until this SOC limit
inverter:
  type: fronius_gen24 #currently only fronius_gen24 supported
  address: 192.168.xx.xx # the local IP of your inverter. needs to be reachable from the machine that runs batcontrol
  user: technician #customer or technician lowercase only!!
  password: *******
  max_charge_rate: 8000 # Watt
  max_grid_power: 15000 # Watt
utility:
  type: awattar_at # [tibber, awattar_at, awattar_de]
#  apikey: YOUR-PASSWORD # only required for tibber get one from https://developer.tibber.com/ Zz-XXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXXx
  vat: 0.20 # only required for awattar
  fees: 0.015 # only required for awattar
  markup: 0.03 # only required for awattar
#  url: http://evcc.local/api/tariff/grid # only required for evcc
  
pvinstallations:
  - name: DachOst #name
    lat: 48.xxx
    lon: 14.xxx
    declination: 10 #inclination toward horizon 0..90 0=flat 90=vertical (e.g. wallmounted)
    azimuth: -90 # -90:East, 0:South, 90:West -180..180
    kWp: 4.000 # power in kWp
    api: #fcsolarapi
  - name: DachWest  #... further installations
    lat: 48.xxx
    lon: 14.xxx
    declination: 10
    azimuth: 90
    kWp: 7.600
    api: #fcsolarapi
consumption_forecast: 
  annual_consumption: 4500 # total consumption in kWh p.a. the load profile 
  load_profile: load_profile.csv #name of the load profile file within the config folder
```

thomeyer1, 2024-11-07:
Ich habe bei mir jetzt in der Konfig den Jahresverbrauch auf 14000 hochgesetzt (Luftwärmepumpe 8KW, die braucht ja alleine schon ca 6-9 MWh im Jahr, denke ich). 14000 ist vlt etwas viel, aber bevor ich das loadprofile anpasse ....

gjuecat, 2024-11-07:
@johannesghd Das ist leider dem Nebel geschuldet. Wetter wäre schön aber ganztags Nebel => der FCST kann somit nicht erreicht werden.
Je später es in den Nachmittag geht, umso weniger kann die Rechnung ja aufgehen => FCST wird auch weniger, Lastprofil am Abend verlangt auch mehr, Batterie ist immer noch leer und die Preisdifferenz zwischen Preis 15 Uhr und Preis 17 Uhr ist immer noch >19ct.
Sollte da dann nicht noch versucht werden nachzuladen?

MaStr, 2024-11-07:
Grundprinzipiell ja.
Konkret müsstest du noch dein logfile posten mit den Werten. Wenn dein Forecast für PV trotzdem mehr als die erwartete Last wäre, würde nix geladen werden.

Zudem gibt es noch von @johannesghd eine Anpassung bei der Ladeberechnung. Da hätte ich gerne noch ein Review vom owner @muexxl . Vielleicht schaffen wir bis Ende der Woche nochmal eine neue Version inklusive HA Version. 

Johannes hat viel Energie investiert hier einige Macken zu finden und aufzubügeln.

gjuecat, 2024-11-07:
[batcontrol.log](https://github.com/user-attachments/files/17667803/batcontrol.log)

MaStr, 2024-11-07:
Ergänzend noch das hier:

```
Available in Battery: 497.6Wh
```

Auf dem Stand bleibt er stehen und verbraucht die Batterie nicht weiter.
Kann es sein, dass du eine Inselschaltung oder eine Notstromdose hast? 
Falls ja, batcontrol nutzt, je nach Version, den falschen Minimum wert für den Akku. Dann stimmt die Berechnung auch nicht ganz.

gjuecat, 2024-11-07:
@johannesghd schaut so aus, als wäre da eine kurze Nachladung gewesen
[image]

gjuecat, 2024-11-07:
@MaStr im Gen24 stehen die SoC Grenzen auf AUTO, unter 7% wird der Speicher aber nie entladen. Ich habe eine manuelle Umschaltung auf einen Inselbetrieb (vom Elektriker selbst erstellt, keine fertige Fronius & CO Lösung).

thomeyer1, 2024-11-07:
> @johannesghd schaut so aus, als wäre da eine kurze Nachladung gewesen ![Screenshot_20241107-194915](https://private-user-images.githubusercontent.com/24219395/384099334-78de6dfa-776d-45b7-a3d8-78311faf87ac.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzEwMDU5NTcsIm5iZiI6MTczMTAwNTY1NywicGF0aCI6Ii8yNDIxOTM5NS8zODQwOTkzMzQtNzhkZTZkZmEtNzc2ZC00NWI3LWEzZDgtNzgzMTFmYWY4N2FjLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDExMDclMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQxMTA3VDE4NTQxN1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTA5NzI5ZmVjYTYyNGU0N2QwOGVhZGQ4YTc5OTkwZGY5NjZkZjQwY2RiNjI1MzhmOGMxNWFmZWY1ZjEwNjIzNmUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.W7-rcMRFXytjvCkyfKvK79ahDxbG6YReGJkT-MhU56M)

so verhielt sich die Anwendung auch bei mir, bis ich den Annual Consumtion Wert hochgesetzt habe
die Anwendung erkennt, dass Strom benötigt wird, aber eben deutlich zu wenig. Die Standard 4500 sind zumindest bei mir viel zu wenig. Heute hate er dann zum Abend hin gut geladen und ich konnte bis 18:45 vom Akku leben. Alles weitere sind sicher Load profile Anpassungen

thomeyer1, 2024-11-07:
habe aber auch die vorgeschlagenen Änderungen von **johannesghd** eingebaut!

MaStr, 2024-11-07:
> @MaStr im Gen24 stehen die SoC Grenzen auf AUTO, unter 7% wird der Speicher aber nie entladen. Ich habe eine manuelle Umschaltung auf einen Inselbetrieb (vom Elektriker selbst erstellt, keine fertige Fronius & CO Lösung).

Dann sollte er aber bis auf `available energy 0W' heruntergehen. Komisch. Eventuell ist der Wert "AUTO" ein anderes Config Feld bei MIN_SOC . @johannesghd  hast du da eine Idee? Hilft ein Auszug der gesicherten Dateien? 
Soll @gjuecat  mal fixe Werte eintragen. 
Weil eigentlich wird bei Available Energy ja MIN_SOC von der Capacity abgezogen.

gjuecat, 2024-11-07:
Ich habe die paar Zeilen in batcontrol.py mal angepasst und werde das weiter beobachten.

Ach ja - nein, nicht im Home Assistant, habe ich aktuell auf einem virtuellen Debian Server laufen
Docker wäre auch schön, hab da schon einige andere Container auch drauf.

normanSp1606, 2024-11-07:
Hallo zusammen,
Bei mir sind es wohl aktuell noch 2 Probleme, die die korrekte Funktionsweise beeinträchtigen.
1. Das Lastprofil: ich nehme das Standardprofil und setze den jährlichen Verbrauch auf 4500 kWh, was derzeit eigentlich mehr als ausreichend sein sollte.
Das es nicht 100% zum aktuellen Nutzungsverhalten passt, geschenkt... heute nachmittag lag der tatsächliche Verbrauch ca. 50% über dem Lastprofil.
Ich versuche als schnelle Lösung dafür einfach den Jahresverbrauch hochzusetzen. Bspw. 9000 kWh, da hat man erstmal genug Puffer.

2. Der PV-Forecast
Die prognostizierte Erzeugung ist ca. doppelt so hoch, wie die tatsächliche. Da es bei mir nur bewölkt und kein Nebel ist, sollte bei korrekter Funktionsweise die Prognose nicht soweit daneben liegen.
Auch ist die Differenz nicht nur zu bestimmten Zeitpunkten, sondern beständig über den ganzen Tag hinweg. Leider bekomme ich die Einzelwerte aus der Fronius Web-UI nicht ohne Premium raus. 
In Home Assistant nutze ich die gleichen Werte für Lat, Long, Az, Dec, und kwp; und auch dort sind die Prognosen zu hoch. Für heute war die Erzeugung-Prognose 4,6kWh, tatsächlich aber nur 2,3kWh.
Da weiß ich nicht, ob ich die forecast.solar-API richtig nutze. Ich verwende aktuell die freie Version und in der Konfig den Parameter
 -   api: https://api.forecast.solar/estimate/51.99/11.99/20/0/5.4
Ist das erstmal soweit korrekt? Stutzig macht mich, das es genau Faktor 2 ist. 
Habt ihr da eine Idee?
Trick 17 wäre: ich setze die kWp auf 2,7. Fühlt sich aber irgendwie falsch an :D

MaStr, 2024-11-07:
> > Dann sollte er aber bis auf `available energy 0W' heruntergehen. Komisch. Eventuell ist der Wert "AUTO" ein anderes Config Feld bei MIN_SOC . @johannesghd hast du da eine Idee? Hilft ein Auszug der gesicherten Dateien?
> 
> Bei welchem timestamp ist das?
> 
> @gjuecat Wann war denn dein Akku das letzte mal über längere Zeit voll?

```$ grep Available Putty.batcontrol.log 
2024-11-07 11:01:49 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 663.5Wh
2024-11-07 11:03:49 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 680.1Wh
2024-11-07 11:05:49 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 713.3Wh
2024-11-07 11:07:49 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 729.9Wh
2024-11-07 11:09:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:11:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:13:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 11:15:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 796.2Wh
2024-11-07 11:17:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 812.8Wh
2024-11-07 11:19:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 829.4Wh
2024-11-07 11:21:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 829.4Wh
2024-11-07 11:23:50 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 846.0Wh
2024-11-07 11:25:51 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 846.0Wh
2024-11-07 11:27:51 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 829.4Wh
2024-11-07 11:29:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 812.8Wh
2024-11-07 11:31:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 796.2Wh
2024-11-07 11:33:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:35:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:37:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:39:52 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:41:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:43:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:45:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:47:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:49:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:51:55 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:53:56 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:55:56 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 11:57:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 11:59:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:01:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:03:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:05:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:07:58 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:09:59 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:11:59 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:14:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:16:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:18:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:20:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:22:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:24:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:26:00 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 779.6Wh
2024-11-07 12:28:01 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:30:03 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:32:03 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:34:03 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 12:36:04 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 12:38:04 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:40:04 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:42:04 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:44:04 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:46:05 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:48:05 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 763.0Wh
2024-11-07 12:50:05 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
2024-11-07 12:52:05 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 746.5Wh
[... cut ...]
2024-11-07 19:23:22 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
2024-11-07 19:25:22 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
2024-11-07 19:27:23 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
2024-11-07 19:29:24 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
2024-11-07 19:31:24 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
2024-11-07 19:33:24 DEBUG [BatCTRL] Reserved Energy: 0.0 Wh. Available in Battery: 497.6Wh
``

MaStr, 2024-11-07:
> Ach ja - nein, nicht im Home Assistant, habe ich aktuell auf einem virtuellen Debian Server laufen Docker wäre auch schön, hab da schon einige andere Container auch drauf.

Docker Image ist in Arbeit. :+1:  #21

MaStr, 2024-11-07:
> 2. Der PV-Forecast
>        Die prognostizierte Erzeugung ist ca. doppelt so hoch, wie die tatsächliche. Da es bei mir nur bewölkt und kein Nebel ist, sollte bei korrekter Funktionsweise die Prognose [...]

Hast du die anderen Beispiel Einträge rausgenommen?
Hast du irgendwie eine besondere Verschattung?

Die Parameter sind ja 

```
lat=51.99
lon=11.99
dec=20
az=0
kwp=5.4

https://api.forecast.solar/estimate/:lat/:lon/:dec/:az/:kwp
``

Ist die Anlage also nach direkt Süden ausgerichtet mit 20° Neigung?

normanSp1606, 2024-11-07:
> Hast du die anderen Beispiel Einträge rausgenommen? Hast du irgendwie eine besondere Verschattung?
>
> Die Parameter sind ja
>
> ```
> lat=51.99
> lon=11.99
> dec=20
> az=0
> kwp=5.4 [...]

Jep, direkt nach Süden.
Keinerlei Verschattung.
Über Postman an forecast.solar bekomme ich die gleichen Werte. batcontrol-config passt also.
Sollte ich mir Sorgen um meine Module machen?

thomeyer1, 2024-11-08:
> Hallo zusammen, Bei mir sind es wohl aktuell noch 2 Probleme, die die korrekte Funktionsweise beeinträchtigen.
>
> 1. Das Lastprofil: ich nehme das Standardprofil und setze den jährlichen Verbrauch auf [...]

@normanSp1606: Ich hatte beim Verbrauch erst einen Denkfehler, da ich meinen Solar Eigenverbrauch nicht beachtet habe. soweit ich erkannt habe, geht der Algorithmus vom Gesamtverbrauch aus - meine Sicht als Tipp

gjuecat, 2024-11-08:
> Bei welchem timestamp ist das?
>
> @gjuecat Wann war denn dein Akku das letzte mal über längere Zeit voll?

Das ist sicher schon über 1 Monat aus, dass der auf 100% geladen war

MaStr, 2024-11-20:
Ich glaube, dass das grundlegende Problem des Issues hier erledigt ist und mache mal zu.

closed 2024-11-20 by MaStr