MaStr/batcontrol #42: crash when charging should start (line 607)

thomeyer1, 2024-11-06:
the calculated value of charge_rate (line 336) is floating (example: 222.60950216020055)., but the function "set_charge_rate" expects int in line 607 "def api_set_charge_rate(self, charge_rate:int):"
changed line 337: from " self.force_charge(charge_rate)" to "self.force_charge(int(charge_rate))" for testing purpose

works for the moment - changed in the on rasphery Pi installed HomeAssistant Docker container

MaStr, 2024-11-06:
Ok, but on the method "set_charge_rate" the parameter type is not specified.
Method api_set_charge_rate is only for external access via MQTT.

thomeyer1, 2024-11-06:
have only the error, thats why I forced charge rate to int in the funktion call - was only a test, maybe issue #34 helps, also changed

Traceback (most recent call last):
File "/batcontrol/./batcontrol.py", line 647, in
bc.run()
File "/batcontrol/./batcontrol.py", line 309, in run
self.set_wr_parameters(net_consumption, price_dict)
File "/batcontrol/./batcontrol.py", line 337, in set_wr_parameters
self.force_charge(charge_rate)
File "/batcontrol/./batcontrol.py", line 507, in force_charge
self._set_charge_rate(charge_rate)
TypeError: 'int' object is not callable

MaStr, 2024-11-06:
Pretty sure this is the already fixed assignment, which haven't made it to the HA version yet.

closed 2024-11-06 by MaStr

thomeyer1, 2024-11-06:
have seen, the result is the same as my test!

MaStr, 2024-11-06:
I'm sorry for that delay.
We have a lot of movement in the repo recently.. and my adjustments with the MQTT API did introduce some bugs :-( 

I tried to organize a few things.. but HA release always steps behind somehow.

thomeyer1, 2024-11-07:
ok, I start the docker container manual in HA environment. whenn starting within HA console, then all changes will be overwritten
so will wait for the new commited version