MaStr/batcontrol #53: User Frontend

muexxl, 2024-11-14:
Hey Guys, 
I would like to get your perspective on creating a web frontend for batcontrol. 

I do have some experience with fastapi (backend) + vue (frontend).
But I am wondering what would be a good strategy to not maintain two sets of APIs one mqtt api and another 

What are your thoughts for a UI ?

ChiliApple, 2024-11-20:
at least parts of the LOG would be great, displaying the forecast and the relevant parts of the LOG.
Entities of it would be great.

ChiliApple, 2024-11-20:
- SOC
- Current price

works over existing entities, the rest sounds good

ichbindaschaos, 2024-12-26:
Yes i would also appreciate entities for my dashboard a lot :) most important would be the current operational mode but the more entities the better :)

muexxl, 2024-12-28:
There are also other reasons for downtimes, so yes. great idea for a feature.

MaStr, 2026-10-19:
Done with 0.10 (in the pipeline to be released)

closed 2026-10-19 by MaStr