MaStr/batcontrol #75: Evaluation forecast drops suddenly from 26hrs to 7hrs

bjreese79, 2024-12-03:
I wondered why the battery discharge lock was suddenly lifted. According to the protocoll, the forecast time jumped back from 26 hours to just 7 hours within one cycle. This led to a different decision by the algorithm. In both cases, the algorithm made the right decision, but obviously just because the forecast time was shortened.
[image]

MaStr, 2024-12-03:
The evaluation round is diverted by the hour.
The latest evaluation was done at 16:00 , which has different references.

At 15:58 , the current price is 0.39 Cent. Relevant for shorten the timeframe is `current_price - min_price_difference` (min_price_difference from your config). 

At 16:00 , the price is now 0.433 Cent . So the next **recharge possibility** compared to the _current price_ is in 7 hours (at 0.344 Cent?).

Later, with reaching the low price timeframe, it will look into the future using current_price + min_price_difference and try to predict how much energy is needed in the hours we want to charge energy for.

muexxl, 2024-12-03:
Another example, why we should give some explanations about the reasoning of the algo.

bjreese79, 2024-12-03:
Thank you for the explanation. I think I've got it ;-)

closed 2024-12-03 by MaStr