MaStr/batcontrol #84: Price difference in % instead of absolute Cent

ichbindaschaos, 2024-12-16:
in case of hight energy prices the condition with an absolute price difference can cause losses. 
example:
load 10kwh from grid and consume 8 kwh from Battery (20% loss)
10 Cent price difference configured

10 x 20cent = 2 EUR cost
8 x 30cent = 2,40 EUR savings
means 40 cent win

10 x 50cent = 5 EUR cost
8 x 60cent = 4,80 EUR savings
means 20 cent loss

it would be better to configure a % value of the low energy price like 40%

10 x 20cent = 2 EUR cost
8 x 28cent = 2,24 EUR savings
means 24 cent win

10 x 50cent = 5 EUR cost
8 x 70cent = 5,60 EUR savings
means 60 cent win

MaStr, 2024-12-16:
Should this replace the fixed target price operation or should it be a second option?
What do you think?

Currently, the fixed value can be calculated up (current_price + min_diff) and down (current_price-min_diff) .

the calculation above suggests low_price*1.4 (40%) im comparison to the higher price, correct?

MaStr, 2024-12-16:
Yes, moving away from a fixed value to % , we will have a breaking change, but much easier in the implementation.

Having both, gives you a flexibility.. but I would deny the startup if both values are set (somewhat breaking change in HomeAssistant configuration).

I _tend_ to have both, but I know, that I tend to over complicate things. So, I am not sure what to do.

At the end, if we move toward or add a % value, I like to publish that "moving" price difference via the API, that you can see that.

MaStr, 2024-12-16:
> Do we have any idea on how many users we'd impact with a breaking change?
No idea. I know, that there is a wider user basis on the https://www.photovoltaikforum.com/ . I can ask there. 
The breaking change would not be instant. I would increase the major version number for this and the HomeAssistant addon, will (correctly adjusted) complain on the wrong/obsolete configuration.

@muexxl  What are your thoughts about that?

ichbindaschaos, 2024-12-17:
hi, after my first tries with charging and discharging it seems to be around 10% loss but you have to note that there are additional batterie lossen when it is not used over time. 
I would suggest to replace the value instead of using both. Simple code is easier to maintain and i asume everybody who uses this addon at this time is capable of adapting to the new variable. 
If you want you can replace the value in the config during the update process with a default ( 30% or so) and state in the doku that there is a config change with new description. 
Then no installation will brake and the user can easily adapt.

ps. great support and fast response i really like this project :)

MaStr, 2024-12-17:
Another thought:

Should the % model be limited by a minimum amount of price ?
difference minimum 0.03 Euro, or above ( 40% , > 0.04... 0.10 Cent whatever).
And this minimum difference needs to be a user setting, depending of the personal price structure.

Each charge cycle is causing wear on the battery.
In my calculation, I would wait with charging for a difference of 0.15 Euro to compensate the battery wear.

MaStr, 2024-12-17:
I simply multiplied the guaranteed cycles with the capacity and divided it by the invest. (Yes, quite expensive).. and added
Some % charging loss.

Yes, I agree totally with your arguments above.

MaStr, 2024-12-17:
> Which ignores that you will have capacity losses even with zero cycles and that the battery isn't worthless after the guarantee ends.

true. In my pov, after that point, I have a positive leading sign .

My idea was to get rid of the high prices hours… and more of the unreasonable high price hours, like 1 euro per kWh.

yes, I’m aware of the fact of the calendar aging of the cells. I mean, I am the optimistic PV+Battery+ECar evangelist in my friend group.. but this time, I tried a more pessimist calculation 🙈

Ok, so: does a lower sediment for the % make sense?

muexxl, 2024-12-18:
Hey guys,
 thanks for the fruitful discussion, 

here are my 2 cents: 

I see three possible solutions: 
1. absolute difference (offset)
2. relative difference (factor)
3. both, offset + linear

For the sake of not overcomplicating things I started off with option 1. 
Especially the behaviour with prices hovering around 0 cent including positives and negatives made me shy away from implementing option 2 (relative). 

However, there are good arguments for all three of them. 
Since it is not a clear decision and I do see two options : 
1. Do nothing
2. put the decision on the user, accepting the burden of another setting that needs to be understood

muexxl, 2024-12-19:
I like the idea.
To avoid a breaking change lets keep the parameter `min_price_difference` representing the threshold.

MaStr, 2024-12-19:
Makes it sense to use abs() for the current price ?

MaStr, 2024-12-19:
> I like the idea of threshold + relative! To avoid a breaking change lets keep the parameter `min_price_difference` representing the threshold and lets introduce a new parameter for the relative component. how about `min_price_difference_rel`?
> 
> The code would result in something like that: `min_recharge_price_level=current_price + max(min_price_difference, min_price_difference_rel * current_price)`

This solution is charming, because if you set _rel to 0, then you can still operate with a fixed value

bjreese79, 2025-01-02:
For those who are interested: I monitor the battery and inverter losses of my Fronius Gen24 10.0 plus and my 11kwh BYD HVM with the Home Assistant and the sunspec integration. Result: the losses between charged and discharged energy in the battery have leveled off at around 5%. The conversion losses from AC to DC and DC to AC are around 5% in each case, although this value is still provisional at the moment, as I have only recently started evaluating and because the inverter does not work at the optimum operating point in winter due to a lack of load.

commit 2c99e09f (2025-02-02): Dynamic price limit (#117)

Closes #84 
* Refactor: Change private method names to use double underscore for name mangling
* pylint batcontrol improvements
closed 2025-02-02 by MaStr