MaStr/batcontrol #87: battery not discharging

neuhubereco, 2024-12-19:
I am running on HA Plugin. 

Battery is at 100% 
the script allows discharging, but the battery is constantly full at 100 % (for the last couple of days). 
(logfile attached). 

2024-12-19 10:48:00 DEBUG [FC Cons] predicting consumption: [2546.3 3730.9 3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7
 2697.8 1766.8 1596.8 1374.8]
2024-12-19 10:48:00 DEBUG [BatCTRL] Production FCST: [ 940. 1050. 1046.  915.  658.  374.   23.    0.    0.    0.    0.    0.
    0.    0.]
2024-12-19 10:48:00 DEBUG [BatCTRL]
[3d00352b_batcontrol_2024-12-19T09-52-05.893Z.log](https://github.com/user-attachments/files/18195864/3d00352b_batcontrol_2024-12-19T09-52-05.893Z.log)
 Consumption FCST: [2546.3 3730.9 3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7
 2697.8 1766.8 1596.8 1374.8]
2024-12-19 10:48:00 DEBUG [BatCTRL] Net Consumption FCST: [1606.3 2680.9 2320.7 1604.9 1957.8 2085.9 2303.9 4519.2 4230.2 2187.7
 2697.8 1766.8 1596.8 1374.8]
2024-12-19 10:48:00 DEBUG [BatCTRL] Prices: [0.14  0.132 0.127 0.147 0.165 0.175 0.18  0.179 0.175 0.171 0.159 0.148
 0.138 0.138]
2024-12-19 10:48:00 DEBUG [BatCTRL] Battery with 13824 Wh above discharge limit 12441 Wh
2024-12-19 10:48:00 INFO [Rule] Discharge allowed due to always_allow_discharge_limit
2024-12-19 10:48:00 INFO [BatCTRL] Mode: Allow Discharging
2024-12-19 10:48:00 INFO [Main] Next evaluation at 10:51:00. Sleeping for 180 seconds

neuhubereco, 2024-12-19:
yes i did. 
i also rebuilt the add on. 
config check. 
restart of HA
checked coms settings in Fronius. 

no other system active for fronius (only solarweb). 
evcc installed but not active. 
https://github.com/wiggal/GEN24_Ladesteuerung  -installed but not active (server is shutoff). 

[image]

MaStr, 2024-12-19:
the schedule is wrong. It shows „entladeleistung“. If the mode is in „discharge“ there is no time entry.
Can you simply remove it and save?

anything here?
[image]

MaStr, 2024-12-19:
No, the schedule is totally correct.
Very strange . Any error messages somewhere ?
Something on the „ Ei­gen­ver­brauchs-Op­ti­mie­rung“

neuhubereco, 2024-12-19:
Something on the „ Ei­gen­ver­brauchs-Op­ti­mie­rung“ -> is on AUTOMATISCH

MaStr, 2024-12-19:
> Something on the „ Ei­gen­ver­brauchs-Op­ti­mie­rung“ -> is on AUTOMATISCH

Ok, same as on my side.
If you delete the schedule entry, does it start discharging?  Maybe there is Kind of a hiccup inside the inverter

neuhubereco, 2024-12-19:
i'll try

> Maybe, the schedule is wrong. It shows „Ladeleistung“. If the mode is in „discharge“ there is no time entry. Can you simply remove it and save?
> 
> anything here? <img alt="image" width="832" src="https://private-user-images.githubusercontent.com/1036501/397350241-13ef98ec-33c8-4c98-b74a-d891042e5059.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzQ2MDkxODksIm5iZiI6MTczNDYwODg4OSwicGF0aCI6Ii8xMDM2NTAxLzM5NzM1MDI0MS0xM2VmOThlYy0zM2M4LTRjOTgtYjc0YS1kODkxMDQyZTUwNTkucG5nP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MTIxOSUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDEyMTlUMTE0ODA5WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9Nzc5MTUzNjM1OGUyMWY0ZmEyMzUwMmJiMzFiMGQ0OWFhNzg0ZjljZmRkYjZlNTA3YTE3MWI4NzRiNGYzYWZhYiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QifQ.bMvczAxh0CELrhX19PoMgrgbXV-9HTJRvkNgh7rXjYQ">

[image]
nothing specific

neuhubereco, 2024-12-19:
I stopped batcontrol and set to: 
[image]

i restarted the service: 

[image]

2024-12-19 12:54:00 INFO [Main] Next evaluation at 12:57:00. Sleeping for 179 seconds
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped
s6-rc: info: service s6rc-oneshot-runner: starting
s6-rc: info: service s6rc-oneshot-runner successfully started
s6-rc: info: service fix-attrs: starting
s6-rc: info: service fix-attrs successfully started
s6-rc: info: service legacy-cont-init: starting
s6-rc: info: service legacy-cont-init successfully started
s6-rc: info: service legacy-services: starting
s6-rc: info: service legacy-services successfully started
2024-12-19 11:56:45 INFO [Main] Starting Batcontrol
2024-12-19 11:56:45 INFO [Main] No logfile path provided. Proceeding with default logfile path: logs/batcontrol.log
2024-12-19 11:56:45 INFO [Batcontrol] host system time zone was not set. Setting to Europe/Berlin
2024-12-19 12:56:45 INFO [Inverter] loading module 
2024-12-19 12:56:45 INFO [Inverter] Login successful
2024-12-19 12:56:45 INFO [FC Cons] The annual consumption of the applied load profile is 3225.2876977966316 kWh 
2024-12-19 12:56:45 INFO [FC Cons] the hourly values from the load profile are scaled with a factor of 6.20 to match the annual consumption of 20000 kWh
2024-12-19 12:56:45 INFO [Main] MQTT Connection enabled
2024-12-19 12:56:45 INFO [MQTT] loading module 
2024-12-19 12:56:45 INFO [MQTT] Waiting for connection
2024-12-19 12:56:46 INFO [MQTT] Waiting for connection
2024-12-19 12:56:46 INFO [MQTT] Connected with result code 0
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/mode/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/charge_rate/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/always_allow_discharge_limit/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/max_charging_from_grid_limit/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/min_price_difference/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/inverters/0/max_grid_charge_rate/set
2024-12-19 12:56:47 DEBUG [MQTT] Registering callback for house/batcontrol/inverters/0/max_pv_charge_rate/set
2024-12-19 12:56:48 INFO [FCSolar] Requesting Information for PV Installation Haus
2024-12-19 12:56:49 DEBUG [FC Cons] predicting consumption: [3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:56:49 DEBUG [BatCTRL] Production FCST: [1060.  917.  658.  374.   23.    0.    0.    0.    0.    0.    0.    0.]
2024-12-19 12:56:49 DEBUG [BatCTRL] Consumption FCST: [3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:56:49 DEBUG [BatCTRL] Net Consumption FCST: [2306.7 1602.9 1957.8 2085.9 2303.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:56:49 DEBUG [BatCTRL] Prices: [0.127 0.147 0.165 0.175 0.18  0.179 0.175 0.171 0.159 0.148 0.138 0.138]
2024-12-19 12:56:49 DEBUG [BatCTRL] Battery with 13824 Wh above discharge limit 12441 Wh
2024-12-19 12:56:49 INFO [Rule] Discharge allowed due to always_allow_discharge_limit
2024-12-19 12:56:49 INFO [BatCTRL] Mode: Allow Discharging
2024-12-19 12:56:49 INFO [Main] Next evaluation at 12:57:00. Sleeping for 11 seconds
2024-12-19 12:57:00 DEBUG [FC Cons] predicting consumption: [3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:57:00 DEBUG [BatCTRL] Production FCST: [1060.  917.  658.  374.   23.    0.    0.    0.    0.    0.    0.    0.]
2024-12-19 12:57:00 DEBUG [BatCTRL] Consumption FCST: [3366.7 2519.9 2615.8 2459.9 2326.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:57:00 DEBUG [BatCTRL] Net Consumption FCST: [2306.7 1602.9 1957.8 2085.9 2303.9 4519.2 4230.2 2187.7 2697.8 1766.8
 1596.8 1374.8]
2024-12-19 12:57:00 DEBUG [BatCTRL] Prices: [0.127 0.147 0.165 0.175 0.18  0.179 0.175 0.171 0.159 0.148 0.138 0.138]
2024-12-19 12:57:00 DEBUG [BatCTRL] Battery with 13824 Wh above discharge limit 12441 Wh
2024-12-19 12:57:00 INFO [Rule] Discharge allowed due to always_allow_discharge_limit
2024-12-19 12:57:00 INFO [BatCTRL] Mode: Allow Discharging
2024-12-19 12:57:00 INFO [Main] Next evaluation at 13:00:00. Sleeping for 179 seconds

MaStr, 2024-12-19:
Does it discharge?

neuhubereco, 2024-12-19:
no - Erzwungene Nachladung

neuhubereco, 2024-12-19:
[image]
yes - thats true..

MaStr, 2024-12-19:
I don’t recommend setting charge_from_grid and always_discharge to the same value. Try 0.90 for charge from grid. Which will prevent bouncing between both actions.

neuhubereco, 2024-12-19:
> Also switch off Modbus if you don't need it prior to restarting.
> 
> I don't recommend using Modbus unless you really know what you're doing as it disables a lot of the internal inverter logic.

hi - no a good idea. Modbus is required for coms between BYD, Smart Meter and Gen24.

MaStr, 2024-12-19:
@johannesghd is talking about modbus/tcp

neuhubereco, 2024-12-19:
i restarted everything, BYD, GEN24, script etc. Switched off Modbus TCP. i am now discharging. 
I'll monitor. If i have troubles again, I'll repost. 
thanks for the support!

closed 2024-12-19 by MaStr