MaStr/batcontrol #95: Wie Grafana Dashboard anzeigen?

WolfImBusch, 2024-12-22:
Einen schönen gute Abend an Alle! Erstmal tolle Software!!!!! 

Ich bitte euch um eure Hilfe! Ich habe heute batcontrol auf meiner Synology als Docker erfolgreich in Betrieb genommen. Mal sehen ob er heute Nacht die Batterie lädt ... 😬 

Gesehen habe ich, dass es auch ein Grafana Dashbord geben soll. Wie kann ich das in Betrieb nehmen? Muss ich die Werte nicht zuerst in ne influxdb schreiben? Ich habe leider keine Dokumentation dazu gefunden .... Ich danke euch und viele Grüße und - schöne Weihnachten!!!

MaStr, 2024-12-22:
Hi,
Ja, es fehlt derweil noch eine Dokumentation dazu. Tut mir leid.
Das ist korrekt mit der influxDB. Der Weg geht über mqtt via Telegraf nach influxDB. 

Ich hatter hier schon mal das zusammengeschrieben: https://www.photovoltaikforum.com/thread/216690-batcontrol-gen24-batterie-tibber-geld-sparen/?postID=4036378#post4036378

WolfImBusch, 2024-12-22:
Ich danke Dir!! Das bekomme ich hin! Viele Grüße!!!!

closed 2024-12-23 by MaStr