MaStr/batcontrol #97: Erforderliche evcc Einstellungen, damit batcontrol das Laden der Hausbatterie steuern kann

WolfImBusch, 2024-12-24:
Hallo liebe Community,

ich habe batcontrol auf meiner Synology als Docker installiert und laut log File sieht auch wirklich alles hervorragend aus! Leider startet das Laden der Hausbatterie nur sehr kurz und bricht dann wieder ab (ich habe das PV Dashboard "Solectrus" installiert - da sieht man das sehr gut). Laut log File gibt batcontrol eindeutig den Befehl zum Laden. Ich habe evcc unter Verdacht, das dagegen zu regeln scheint. 

Mir ist nicht so ganz klar, welche Einstellungen ich in evcc vornehmen muss, so dass sich evcc nicht mehr um das Laden der Batterie kümmert, sondern batcontrol den Job überlässt. Versucht habe ich in evcc
a) Hausbatterie- --> Netzladen --> kein Limit
b) Hausbatterie --> Batterienutzung --> Verhindere Entladung im Schnell-Modus und bei geplantem Laden --> deaktiviert
c) Konfiguration --> experimentelle Features --> deaktiviert
d) in dem evcc.yaml die Steuerung des Fronius wieder zurückgedreht von dem Template "fronius-gen24" auf "fronius-solarapi-v1"

Hat noch jemand eine Idee? (Wäre vielleicht auch was für die Doku)

Helfen würde mir beim Finden/Eingrenzen der Ursache auch, wenn ich batcontrol testweise irgendwie dazu bringen könnte, den Ladevorgang zu starten (min_price_difference habe ich schon auf 0.01 gesetzt) während der evcc Container gestoppt ist. Gibt es da eine Möglichkeit?

Herzlichen Dank an Alle und viele Grüße!

MaStr, 2024-12-24:
Also wir müssen hier mehrere Dinge sortieren:

- Die evcc Anbindung seitens batcontrol kümmert sich nur um das Sperren der Batterie durch batcontrol, sobald evcc den Ladevorgang des Elektro-Autos startet.
- Wenn man über die Modbus Schnittstelle (modbus/tcp) etwas steuert (z.B. evcc), dann überschreibt es die Zeitpläne, die batcontrol nutzt zur Steuerung. Das bedeutet auch, dass ein Teste mit evcc+modbus/tcp seiteneffekte haben kann. Bei mir läuft evcc mit Modbus/tcp, aber ich habe im evcc die Steuerung ausgestellt (Haken bei Sperren ausgeschaltet).
- Es kann bei „Ladehemmungen“ auch helfen den inverter mal neu zu starten, wenn von diversen Versuchen es zu Probleme. Kommt.
- Wenn du über MQTT mode=-1 das Laden aktivierst, dann wird ein Standard-Wert genommen. Dieser ist nicht die in der batcontrol.yml definierte Obergrenze. Wenn du über mqtt das Zwangsladen testen möchtest, dann verwende am besten /charge_rate/set und setze die gewünschte Watt Zahl ein.
- Im laufenden Betrieb entscheidet batcontrol wie folgt:

1. Ist ein zukünftiger Preis über den aktuellen Wert um min_price_difference ? Ja, dann laden wir.
2. Geladen wird nicht pauschal voll, sondern um den Wert der für die Stunden current_price+min_price_difference in frage kommt. Wenn in dem Zeitraum die Verbrauchsvorhersage sagt, dass du 600Wh brauchst, lädt er ca. 600Wh nach. Mindestens wird mit 500W die Batterie aus dem Netz geladen.

- die Batterie braucht bei maximaler charge werte durchaus einige Minuten bis die maximale Ladeleistung anliegt.
- Das Debuglog sagt, dass in 10 Stunden ein Zeitfenster mit niedrigen Preisen kommt um eventuell nachladen, deswegen lädt er *jetzt* nicht.
- Bei der Uhrzeit musst du als erstes mit dem Synology starten? Welche Uhrzeit hat es, und ganz wichtig: welche Zeitzone?

muexxl, 2024-12-24:
Bzgl der Timestamps: Versuche es mal mit "Europe/Berlin" . Bei dir wird warscheinlich noch UTC Zeit aus dem Dockercontainer genutzt

muexxl, 2024-12-24:
Ansonsten : Versuch einfach mal modbus TCP im Inverter zu deaktivieren und den Inverter dann neuzustarten.

WolfImBusch, 2024-12-24:
Hallo @MaStr und @muexxl ,

zuerst herzlichen Dank für Eure Hilfe und Anmerkungen. 

Ich denke, ich habe die Funktionsweise einigermaßen verstanden. Ich lasse batcontrol mal ein paar Tage laufen und beobachte das/schau mir die Logs an. Momentan würde ich auf einen Neu-Start des Inverters gerne vermeiden .. bei evcc habe ich den Haken zum Sperren ebenfalls entfernt (s. meinen Post oben).

Aktuell/mittlerweise ist batcontrol der Meinung, dass es für sinnvolles Laden kein geeignetes Zeitfenster gibt. Auszug aus dem Log

```
2024-12-24 22:50:49 DEBUG [EVCC] Received message on evcc/loadpoints/1/charging
2024-12-24 22:51:00 DEBUG [FC Cons] predicting consumption: [283.6 255.7 260.5 251.7 245.7 391.1 457.5 395.4 548.7 701.  609.6 586.2
 611.3 926.7 595.8 690.2 576.4 761.4]
2024-12-24 22:51:00 DEBUG [BatCTRL] Production FCST: [  0.   0.   0.   0.   0.   0.   0.   0.   0. 114. 407. 554. 616. 602.
 539. 434. 282.  52.]
2024-12-24 22:51:00 DEBUG [BatCTRL] Consumption FCST: [283.6 255.7 260.5 251.7 245.7 391.1 457.5 395.4 548.7 701.  609.6 586.2
 611.3 926.7 595.8 690.2 576.4 761.4]
2024-12-24 22:51:00 DEBUG [BatCTRL] Net Consumption FCST: [283.6 255.7 260.5 251.7 245.7 391.1 457.5 395.4 548.7 587.  202.6  32.2
  -4.7 324.7  56.8 256.2 294.4 709.4]
2024-12-24 22:51:00 DEBUG [BatCTRL] Prices: [0.314 0.315 0.31  0.308 0.303 0.297 0.303 0.306 0.308 0.314 0.325 0.318
 0.317 0.319 0.315 0.325 0.341 0.347]
2024-12-24 22:51:00 DEBUG [Rule] Evaluating next 18 hours until 16:59
2024-12-24 22:51:00 DEBUG [Rule] Reserved Energy will be used in the next hours: [1, 9, 10, 11, 12, 13, 14, 15, 16, 17]
2024-12-24 22:51:00 DEBUG [Rule] Reserved Energy: 2714.3 Wh. Usable in Battery: 0.0 Wh
2024-12-24 22:51:00 DEBUG [Rule] Discharge forbidden. Stored usable energy 0.0 Wh <= Reserved energy 2714.3 Wh
2024-12-24 22:51:00 DEBUG [Rule] Discharging is NOT allowed
2024-12-24 22:51:00 DEBUG [BatCTRL] Charging allowed: True
2024-12-24 22:51:00 DEBUG [Rule] Charging is allowed, because SOC is below 90%
2024-12-24 22:51:00 DEBUG [Rule] No additional energy required or possible price found.
2024-12-24 22:51:00 INFO [BatCTRL] Mode: Avoid Discharging
2024-12-24 22:51:00 INFO [Main] Next evaluation at 23:54:00. Sleeping for 180 seconds
```

Bezüglich der Zeitzone (ist ja wirklich nicht super wichtig) .. Uhrzeit auf meiner Synology ist Zeitzone Europe/Berlin - das passt. Um den Docker-Container auch in diese Zeitzone zu katapultieren gibt es ja einerseits in der Datei "batcontrol_config.yaml" den Eintrag
`timezone: Europe/Berlin #your time zone. not optional`

Zudem habe ich im Docker noch den Parameter
[image]
übergeben. 

Laut Logfile ist der Container auch der Meinung, dass die Zeitzone die richtige ist
`2024-12-24 12:29:27 INFO [Batcontrol] host system time zone is Europe/Berlin`

Die Uhrzeit im Log-File beeindruckt das aber trotzdem noch nicht ... :-)

Viele Grüße und stressfreie Tage euch!

WolfImBusch, 2024-12-25:
Einen schönen guten Morgen,

oh - ging schneller als gedacht. Seit heute früh um 4 Uhr möchte Batcontrol gerne die Hausbatterie nachladen. 
[image]

Leider startet das Laden nicht. Die Einstellungen im Fronius sehen aber m.E. gut aus
[image]

Auch die MQTT Werte sehen gut aus
[image]

Anbei noch der letzte Log Eintrag ...
```
2024-12-25 07:39:00 DEBUG [FC Cons] predicting consumption: [ 701.   609.6  586.2  611.3  926.7  595.8  690.2  576.4  761.4 1029.5
 1117.3  658.1  555.1  499.6  331.4  282.1]
2024-12-25 07:39:00 DEBUG [BatCTRL] Production FCST: [ 88. 299. 394. 443. 449. 413. 333. 220.  42.   0.   0.   0.   0.   0.
   0.   0.]
2024-12-25 07:39:00 DEBUG [BatCTRL] Consumption FCST: [ 701.   609.6  586.2  611.3  926.7  595.8  690.2  576.4  761.4 1029.5
 1117.3  658.1  555.1  499.6  331.4  282.1]
2024-12-25 07:39:00 DEBUG [BatCTRL] Net Consumption FCST: [ 613.   310.6  192.2  168.3  477.7  182.8  357.2  356.4  719.4 1029.5
 1117.3  658.1  555.1  499.6  331.4  282.1]
2024-12-25 07:39:00 DEBUG [BatCTRL] Prices: [0.314 0.325 0.318 0.317 0.319 0.315 0.325 0.341 0.347 0.35  0.348 0.347
 0.348 0.341 0.342 0.331]
2024-12-25 07:39:00 DEBUG [Rule] Evaluating next 16 hours until 23:59
2024-12-25 07:39:00 DEBUG [Rule] Reserved Energy will be used in the next hours: [1, 2, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12, 13, 14, 15]
2024-12-25 07:39:00 DEBUG [Rule] Reserved Energy: 7237.8 Wh. Usable in Battery: 0.0 Wh
2024-12-25 07:39:00 DEBUG [Rule] Discharge forbidden. Stored usable energy 0.0 Wh <= Reserved energy 7237.8 Wh
2024-12-25 07:39:00 DEBUG [Rule] Discharging is NOT allowed
2024-12-25 07:39:00 DEBUG [Rule] Required Energy: 4079.4 Wh is based on next 'high price' hours [8, 9, 10, 11, 12]
2024-12-25 07:39:00 DEBUG [Rule] Stored usable Energy: 0.0 , Recharge Energy: 4079.4 Wh
2024-12-25 07:39:00 DEBUG [BatCTRL] Charging allowed: True
2024-12-25 07:39:00 DEBUG [Rule] Charging is allowed, because SOC is below 90%
2024-12-25 07:39:00 DEBUG [BatCTRL] Get additional energy via grid: 4079.4 Wh
2024-12-25 07:39:00 INFO [BatCTRL] Mode: grid charging. Charge rate : 4000 W
2024-12-25 07:39:00 INFO [Main] Next evaluation at 08:42:00. Sleeping for 179 seconds
2024-12-25 07:39:49 DEBUG [EVCC] Received message on evcc/loadpoints/1/charging
2024-12-25 07:40:49 DEBUG [EVCC] Received message on evcc/loadpoints/1/charging

```

Habt ihr noch ne Idee? Doch den Wechselrichter neu starten? Hab Bammel ... :-)

Herzlichen Dank und viele Grüße!!!

MaStr, 2024-12-25:
Moin,
also das mit der Zeit ist unter Umständen dennoch ein Problem, weil dann die Batterie falsch freigeschaltet/gesperrt/geladen wird. Vergleiche mal die Reihen der Preise aus dem log mit den aktuellen und zukünftigen Preisen aus der anderen Quelle. Vielleicht ist das log zwar falsch, aber die restlichen Daten korrekt verarbeitet sind. (Es könnte sein, dass das Programm für das Log mit UTC initialisiert wird, deswegen die logzeilen Quatsch sind, aber der Rest ok).

Wir hatten im Photovoltaik Forum das Problem auch schonmal.
Erst ein Update und/oder damit verbundene Neustart hatten das Problem gelöst.

WolfImBusch, 2024-12-25:
Moin!
also das sieht alles sehr gut aus ...

2024-12-25 08:06:00 DEBUG [BatCTRL] Prices: [0.325 0.318 0.317 0.319 0.315 0.325 0.341 0.347 0.35  0.348 0.347 0.348 0.341 0.342 0.331]

==>
0.325 9-10h
0.318 10-11h 
0.317 11-12h 
0.319 12-13h 
0.315 13-14h 
0.325 14-15h
0.341 15-16h

Stichprobe Tibber Preis zwischen 15-16 Uhr sind 34.1 cent
[image]

Ich nehme mir jetzt allen Mut zusammen und starten den Inverter neu :-) Bitte drück die Daumen :-)

WolfImBusch, 2024-12-25:
Hi @MaStr 

der Inverter ist wieder da - Neustart hat geklappt ... :-) Ich schaue mal, wann batcontrol wieder das nächste Mal laden will. Nach Neustart Inverter und Neu-Installation von batcontrol ist er aktuell im Modus "Avoid Discharging" - Preis-Differenz habe ich auf 0.01 gesetzt - mal sehen ...

Was mir noch aufgefallen ist im Log ... (s.u. - zwei letzte Zeile) ... hier gibt es tatsächlich m.E. im Zeitstempel des Logs die falsche Uhrzeit - im Text selbst aber die richtige ...

2024-12-25 **08:42:00** INFO [Main] Next evaluation at **09:45:00**. Sleeping for 180 seconds

Erste Zeilen Log nach Neustart
2024-12-25 08:37:34 INFO [Batcontrol] **host system time zone was not set. Setting to Europe/Berlin**
2024-12-25 **08:37:34** INFO [Inverter] loading module 
2024-12-25 **08:37:34** INFO [Inverter] Login successful

Vielleicht kommt batcontrol tatsächlich irgendwie mit den Uhrzeiten durcheinander ... Sieht aber so aus wie Du geschrieben hast, dass der Zeitstempel sind nach Ändern der Zeitzone nicht ändert und das Log nach UTC initialisiert wird.

Falls ich bei der Suche irgendwie behilflich sein kann, sag bitte Bescheid. 

Viele Grüße!!!

```
2024-12-25 08:41:49 DEBUG [EVCC] Received message on evcc/loadpoints/1/charging
2024-12-25 08:42:00 DEBUG [FC Cons] predicting consumption: [ 609.6  586.2  611.3  926.7  595.8  690.2  576.4  761.4 1029.5 1117.3
  658.1  555.1  499.6  331.4  282.1]
2024-12-25 08:42:00 DEBUG [BatCTRL] Production FCST: [314. 418. 441. 412. 355. 280. 183.  35.   0.   0.   0.   0.   0.   0.
   0.]
2024-12-25 08:42:00 DEBUG [BatCTRL] Consumption FCST: [ 609.6  586.2  611.3  926.7  595.8  690.2  576.4  761.4 1029.5 1117.3
  658.1  555.1  499.6  331.4  282.1]
2024-12-25 08:42:00 DEBUG [BatCTRL] Net Consumption FCST: [ 295.6  168.2  170.3  514.7  240.8  410.2  393.4  726.4 1029.5 1117.3
  658.1  555.1  499.6  331.4  282.1]
2024-12-25 08:42:00 DEBUG [BatCTRL] Prices: [0.325 0.318 0.317 0.319 0.315 0.325 0.341 0.347 0.35  0.348 0.347 0.348
 0.341 0.342 0.331]
2024-12-25 08:42:00 DEBUG [Rule] Evaluating next 15 hours until 23:59
2024-12-25 08:42:00 DEBUG [Rule] Reserved Energy will be used in the next hours: [6, 7, 8, 9, 10, 11, 12, 13, 14]
2024-12-25 08:42:00 DEBUG [Rule] Reserved Energy: 5593.0 Wh. Usable in Battery: 0.0 Wh
2024-12-25 08:42:00 DEBUG [Rule] Discharge forbidden. Stored usable energy 0.0 Wh <= Reserved energy 5593.0 Wh
2024-12-25 08:42:00 DEBUG [Rule] Discharging is NOT allowed
2024-12-25 08:42:00 DEBUG [BatCTRL] Charging allowed: True
2024-12-25 08:42:00 DEBUG [Rule] Charging is allowed, because SOC is below 90%
2024-12-25 08:42:00 DEBUG [Rule] No additional energy required or possible price found.
2024-12-25 08:42:00 INFO [BatCTRL] Mode: Avoid Discharging
2024-12-25 08:42:00 INFO [Main] Next evaluation at 09:45:00. Sleeping for 180 seconds
2024-12-25 08:42:49 DEBUG [EVCC] Received message on evcc/loadpoints/1/charging
```

MaStr, 2024-12-25:
Du kannst ja versuchen über das charge_rate/set für ein paar Minuten mal das Laden zu starten

WolfImBusch, 2024-12-25:
Funktioniert super ... 

[image]

[image]

MaStr, 2024-12-25:
Yay 👍🚀

WolfImBusch, 2024-12-25:
Wenn es jetzt batcontrol machen würde und ich nicht manuell wäre das super klasse ... :-)

WolfImBusch, 2024-12-25:
**ER LÄDT!!! ER LÄDT!! Schönstes Weihnachtsgeschenk überhaupt!!!!!! :-)**

Seit 13 Uhr lädt er - nachdem batcontrol vermutlich die Preise von Tibber für den nächsten Tag erhalten hat!!!

Was letztendlich zum Ziel geführt hat - ich kann es nicht genau sagen. Ich gehe mal ganz stark davon aus, dass es der Inverter Reset war. Ich hatte Hoffnung, nachdem das manuelle Setzen der charge_rate per MQTT dazu geführt hat, dass der Haus-Akku mit 3 kW geladen wurde und nicht wie bei meinem ersten Post nur mit 800 W. Kann sein, dass evcc bei meinen Batterie-Lade Tests irgendwelche Register gesetzt hat. Wird wohl für immer Geheimnis bleiben ... :-)

Nochmal vielen herzlichen Dank für eure Anmerkungen und Hilfe und - natürlich für eure Geduld!!!!

Viele Grüße!!!!

PS: Das Thema UTC Zeitstempel im Logfile (s.a. #98 ) scheint ein weit verbreitetes Problem zu sein. Das Internet ist quasi voll damit ... :-)

MaStr, 2024-12-25:
:-D 

freut mich, frohe Weihnachten :-)

closed 2024-12-25 by MaStr