MaStr/batcontrol #99: EVCC ValueError: too many values to unpack (expected 2)when parsing load points from MQTT

Ak11rA, 2024-12-27:
I'm trying to run BatControl with the EVCC connection:

Config is:

```
evcc:
  enabled: true
  broker: localhost
  port: 1883
  status_topic: evcc/status
  loadpoint_topic:
    - evcc/loadpoints/1/charging
    - evcc/loadpoints/2/charging
  tls: false
```

EVCC data in MQTT looks good to me:

[image]

But something seems to be wrong with parsing the load points data. 
I'm getting the error:

`ValueError: too many values to unpack (expected 2)
`

[image]

Any ideas?

I'm using the latest docker image for BatControl.

MaStr, 2024-12-27:
Latest is the last release and does support only one loadpoint.

the multiple loadpoint version is not released yet. I can build an intermediate testing image on the current development branch „main“.

MaStr, 2024-12-27:
Try using tag „main“ instead of latest. This was just built and is like a snapshot build. That should work.

Ak11rA, 2024-12-27:
Thanks, the main docker tag works! Very promising project.

closed 2024-12-27 by Ak11rA