MaStr/batcontrol #101: Standby function

bjreese79, 2025-01-01:
Switching the inverter to standby during periods without charging, discharging or PV production. The mode can be activated (force standby) and deactivated again via the Fronius Web UI. This makes sense as the consumption of the inverter in standby is around 50W. The inverter sometimes switches to standby itself, but the conditions for this are unclear. I have sometimes observed that it does not switch to standby at all throughout the night.

bjreese79, 2025-01-01:
I found something in the operating instructions about the conditions for activating and deactivating standby mode. Apparently the inverter only switches to standby if one of the conditions is that the battery is at minimum charge. This means that it never switches to standby if batcontrol is holding back charge overnight for the next day. This results in the inverter remaining in operation all night and consuming power.[image]

Ak11rA, 2025-01-02:
I think sometimes the inverter also doesn't charge from grid even if configured by batcontrol when the battery is in standby mode.

bjreese79, 2025-01-02:
Possibly yes, but I did not observe such behavior at all. I think the standby behavior should be controlled by batcontrol anyway. Maybe it is sufficient to just set the min soc at the current level, when discharging has to be avoided. This would give the inverter the possibility to switch into standby by its own. Once discharging is allowed again or charging from grid needs to be activated the min soc will be set to its original value (i.e. 5%) to get the inverter out of standby again.

bjreese79, 2025-01-05:
My inverter measures 35-40W in standby mode. That can easily add up to 0.5kwh in one night. You can now dismiss this as peanuts, but taking into account the losses of the inverter, the entire principle of grid charging is based on savings of cents or a few euros per day. I therefore think it is quite relevant. And if something like this is easy to implement, why should we not do it?

bjreese79, 2025-01-05:
I measured 40 watts between standby mode (fronius energy-saving mode) and an active inverter that is only supposed to maintain a certain state of charge of the battery, but does not perform any feeding, PV production, charging or discharging (see 2 different screenshots). And actually, my aim is to follow the internal logic of the inverter, which normally switches to energy-saving mode when not in use. This seems to be prevented by the current batcontrol logic. I don't want to complain here, but I am totally enthusiastic about batcontrol. Unfortunately, I cannot assess how complex such an implementation would be. Perhaps there will be a statement from the developers. [image][image]

bjreese79, 2025-01-05:
I use the sunspec integration in the Home Assistant to read out the measured values of the inverter directly via modbus. I am not claiming that the inverter is running with a total of 0 watts in energy-saving mode. But on one measured phase (presumably AC) there is always at least 40W during operation and 0 in energy-saving mode. The other phase measures the DC power. Either the converted value or the value to be converted. Both values increase when AC is converted to DC or vice versa. According to my interpretation, the difference between the two phases should correspond to the conversion loss.

bjreese79, 2025-01-06:
By the way this is obviously the reason why the fronius switches to energy-saving mode below 50W PV power, as this corresponds to self-consumption (when batt is at min. SOC at the same time).

MaStr, 2025-01-06:
@bjreese79 , @johannesghd is a contributor with a huge amount of experience in the Fronius Inverters.

If we can save 50w per hour, we can save 50 Watts.
I’m wondering if that is running in contradiction against our 3 minute check against multiple parameters.
I don’t know what will happen if we are putting the inverter into the standby while we are trying to ask for SOC 3 minutes laters. And how to wake it up, if we want to switch a mode or do something else.

bjreese79, 2025-01-06:
You are obviously right. Taking the ac_meter value into account which shows the power I am consuming from the grid, it is only jumping by approximately 20W when the inverter switches to the energy safing mode (see screenshot). In this case I can follow your argumentation, that the losses are less than I originally expected. Thank you for the discussion. 
[image]

MaStr, 2025-11-11:
I created a monitoring on my end to look at the inverter state.
As we changed the algo to only send changes if there *is* a change, it might helpt to push the inverter into sleep state