MaStr/batcontrol #113: invalid login 01:00:00 Am after installing the latest HA Version

thomeyer1, 2025-01-18:
HI,

after installing the last HA version of batcontrol, i got very often exactly 01:00:00 Am the same error:

2025-01-18 00:57:00 DEBUG [BatCTRL] Charging allowed: True
2025-01-18 00:57:00 DEBUG [Rule] Charging is allowed, because SOC is below 75%
2025-01-18 00:57:00 DEBUG [Rule] No additional energy required or possible price found.
2025-01-18 00:57:00 INFO [BatCTRL] Mode: Avoid Discharging
2025-01-18 00:57:00 INFO [Inverter] Login successful
2025-01-18 00:57:00 INFO [Main] Next evaluation at 01:00:00. Sleeping for 180 seconds
2025-01-18 01:00:00 DEBUG [FC Cons] predicting consumption: [ 535.1  596.1  547.7  539.5  568.1  594.3  877.7 1340.  1733.3 2314.7
  946.3 1051.5  997.6 1056.3 1812.7 1073.5  757.3 1124.5  826.2  988.8
 1008.1  785.   516.3]
2025-01-18 01:00:00 DEBUG [BatCTRL] Production FCST: [   0.    0.    0.    0.    0.    0.    0.  534. 1663. 2409. 2567. 2176.
 1539.  961.  577.  131.    0.    0.    0.    0.    0.    0.    0.]
2025-01-18 01:00:00 DEBUG [BatCTRL] Consumption FCST: [ 535.1  596.1  547.7  539.5  568.1  594.3  877.7 1340.  1733.3 2314.7
  946.3 1051.5  997.6 1056.3 1812.7 1073.5  757.3 1124.5  826.2  988.8
 1008.1  785.   516.3]
2025-01-18 01:00:00 DEBUG [BatCTRL] Net Consumption FCST: [  535.1   596.1   547.7   539.5   568.1   594.3   877.7   806.     70.3
   -94.3 -1620.7 -1124.5  -541.4    95.3  1235.7   942.5   757.3  1124.5
   826.2   988.8  1008.1   785.    516.3]
2025-01-18 01:00:00 DEBUG [BatCTRL] Prices: [0.317 0.317 0.315 0.312 0.316 0.316 0.332 0.346 0.345 0.329 0.318 0.315
 0.31  0.315 0.332 0.351 0.369 0.372 0.373 0.359 0.344 0.335 0.327]
2025-01-18 01:00:00 DEBUG [Rule] Evaluating next 23 hours until 23:59
2025-01-18 01:00:00 DEBUG [Rule] Reserved Energy will be used in the next hours: [6, 7, 8, 9, 10, 14, 15, 16, 17, 18, 19, 20, 21, 22]
2025-01-18 01:00:00 DEBUG [Rule] Reserved Energy: 6557.7 Wh. Usable in Battery: 0.0 Wh
2025-01-18 01:00:00 DEBUG [Rule] Discharge forbidden. Stored usable energy 0.0 Wh <= Reserved energy 6557.7 Wh
2025-01-18 01:00:00 DEBUG [Rule] Discharging is NOT allowed
2025-01-18 01:00:00 DEBUG [BatCTRL] Charging allowed: True
2025-01-18 01:00:00 DEBUG [Rule] Charging is allowed, because SOC is below 75%
2025-01-18 01:00:00 DEBUG [Rule] No additional energy required or possible price found.
2025-01-18 01:00:00 INFO [BatCTRL] Mode: Avoid Discharging
2025-01-18 01:00:00 INFO [Inverter] Login failed 3 times .. aborting
2025-01-18 01:00:00 INFO [Main] Shutting down Batcontrol
2025-01-18 01:00:00 INFO [Inverter] Reverting batcontrol created config changes
2025-01-18 01:00:00 INFO [Inverter] Restoring previous battery configuration: {"BAT_M0_SOC_MAX": 100, "BAT_M0_SOC_MIN": 5, "BAT_M0_SOC_MODE": "auto", "HYB_BM_CHARGEFROMAC": false, "HYB_EM_MODE": 0, "HYB_EM_POWER": 0, "HYB_EVU_CHARGEFROMGRID": false} 
2025-01-18 01:00:00 INFO [Inverter] Login failed 3 times .. aborting
Traceback (most recent call last):
  File "/batcontrol/./batcontrol.py", line 896, in <module>
    bc.run()
  File "/batcontrol/./batcontrol.py", line 427, in run
    self.set_wr_parameters(net_consumption, price_dict)
  File "/batcontrol/./batcontrol.py", line 479, in set_wr_parameters
    self.avoid_discharging()
  File "/batcontrol/./batcontrol.py", line 707, in avoid_discharging
    self.inverter.set_mode_avoid_discharge()
  File "/batcontrol/inverter/fronius.py", line 279, in set_mode_avoid_discharge
    return self.set_time_of_use(timeofuselist)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 353, in set_time_of_use
    response = self.send_request(
               ^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 400, in send_request
    response = self.login()
               ^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 421, in login
    return self.send_request(path, auth=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 400, in send_request
    response = self.login()
               ^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 421, in login
    return self.send_request(path, auth=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 400, in send_request
    response = self.login()
               ^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 421, in login
    return self.send_request(path, auth=True)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/batcontrol/inverter/fronius.py", line 398, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Login failed repeatedly .. wrong credentials?

what is going wrong, any idea?

MaStr, 2025-01-18:
Not your fault.
This happens on my side occasionally.

I’ll try to add some code, which pauses re-login attempts for 30 seconds in the case of not being the first login.

Currently, It is not clear what is the root cause.

thomeyer1, 2025-01-18:
Hi Matthias,

Than was my idea - weit some time before Next Login. Habe checked my Environment, and the 5g internetrouter can reboot every 7days between 1 and 5 am. Not sure, if this is the root cause. The error occurs after 3, 5 or some time every day. So, will wait for the code.

Herzliche Grüße
Thomas Meyer

MaStr, 2025-01-18:
It’s not the root cause.
It’s always at 1 o’clock… my DSL has different disconnect times.

Must be.some kind of Fronius internal stuff

MaStr, 2025-01-18:
I already added a more detailed error message with 5cba89b and with 
c603e4e60a0a6b06455b55f37b98e116a57ef10a there is now a 10 second delay on logins after the first login (fail fast on startup, but fail slow during runtime)

closed 2025-01-18 by MaStr