MaStr/batcontrol #118: logfile is overwritten on restart of batcontrol

muexxl, 2025-01-23:
The logfile is overwritten when the script restarts. 
This makes it difficult to analyze logs in a containerized setup. 

Potential solution: 
open logfile in append mode

MaStr, 2025-01-23:
A container consists of two layers:

- image
- overlay layer

All writes are written to the overlay layer, the original image is not changed.

If you create a container, stop and start it, the overlay layer stays. As soon as a container is removed and newly created, all changes in the overlay layer are gone.

From what I know, the script is already opening the logfile in append mode.

muexxl, 2025-01-23:
logfile is now created at /data/batcontrol.log and a symlink is placed in /app/config/logs/
Thereby, the logs are persistent

closed 2025-01-23 by muexxl

MaStr, 2025-01-24:
`/app/config/logs/` not `/app/logs/` ?