MaStr/batcontrol #120: #113 Bug still exists after ubgrade do newest version

thomeyer1, 2025-01-31:


MaStr, 2025-01-31:
Correct, found it . First fix is in 0100-bug branch.

But I need to refactor the send_method

MaStr, 2025-01-31:
#113

MaStr, 2025-02-11:
```
2025-02-11 01:00:01 INFO [BatCTRL] Mode: Avoid Discharging
2025-02-11 01:00:01 INFO [Main] Shutting down Batcontrol
2025-02-11 01:00:01 INFO [Inverter] Reverting batcontrol created config changes
2025-02-11 01:00:01 INFO [Inverter] Restoring previous battery configuration: {"BAT_M0_SOC_MAX": 100, "BAT_M0_SOC_MIN": 5, "BAT_M0_SOC_MODE": "auto", "HYB_BM_CHARGEFROMAC": true, "HYB_EM_MODE": 0, "HYB_EM_POWER": -50, "HYB_EVU_CHARGEFROMGRID": true}
2025-02-11 01:00:01 ERROR [Inverter] Login -0- failed, Response: <Response [401]>
2025-02-11 01:00:01 ERROR [Inverter] Response-raw: <urllib3.response.HTTPResponse object at 0xb2b79af0>
2025-02-11 01:00:01 INFO [Inverter] Retrying login in 10 seconds
2025-02-11 01:00:11 INFO [Inverter] Login successful <Response [200]>
2025-02-11 01:00:11 INFO [Inverter] Logout successful
Traceback (most recent call last):
  File "/home/matze/batcontrol/batcontrol.py", line 1074, in <module>
    bc.run()
  File "/home/matze/batcontrol/batcontrol.py", line 519, in run
    self.set_wr_parameters(net_consumption, price_dict)
  File "/home/matze/batcontrol/batcontrol.py", line 576, in set_wr_parameters
    self.avoid_discharging()
  File "/home/matze/batcontrol/batcontrol.py", line 844, in avoid_discharging
    self.inverter.set_mode_avoid_discharge()
  File "/home/matze/batcontrol/inverter/fronius.py", line 343, in set_mode_avoid_discharge
    return self.set_time_of_use(timeofuselist)
  File "/home/matze/batcontrol/inverter/fronius.py", line 434, in set_time_of_use
    response = self.send_request(
  File "/home/matze/batcontrol/inverter/fronius.py", line 481, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Request 0 failed with 403-Forbidden.
         path:/config/timeofuse,
        params:None
        headers {}
        nonce 51a0678d67aa924d
        payload {"timeofuse": [{"Active": true, "Power": 0, "ScheduleType": "DISCHARGE_MAX", "TimeTable": {"Start": "00:00", "End": "23:59"}, "Weekdays": {"Mon": true, "Tue": true, "Wed": true, "Thu": true, "Fri": true, "Sat": true, "Sun": true}}]}

```
This is happening at 01.00 o'clock.
Still not clear what is wrong.