MaStr/batcontrol #125: wahrscheinlich seit FroniusGen24 Firmwareupdate auf 1.35.4-1

danielaussieb, 2025-02-10:
bekomme das BATcontrol offenbar seit dem heutigen Firmwareupdate auf 1.35.4-1 des GEN24 nicht mehr zu laufen
bin selbst überhaupt kein Entwickler nur interessierter und begeisterter Anwender

folgende LOGdaten

2025-02-10 12:29:57 INFO [Main] Starting Batcontrol
2025-02-10 12:29:57 INFO [Main] No logfile path provided. Proceeding with default logfile path: logs/batcontrol.log
2025-02-10 12:29:57 INFO [Batcontrol] host system time zone was not set. Setting to Europe/Berlin
2025-02-10 13:29:57 INFO [Inverter] loading module 
2025-02-10 13:29:57 INFO [Inverter] Login successful
2025-02-10 13:29:57 INFO [Inverter] Login successful
2025-02-10 13:29:57 INFO [Inverter] Login successful
2025-02-10 13:29:57 ERROR [Inverter] Failed to get battery configuration. Returning empty dict
2025-02-10 13:29:57 INFO [Inverter] Login successful
2025-02-10 13:29:57 INFO [Inverter] Login successful
2025-02-10 13:29:57 INFO [Inverter] Login successful
Traceback (most recent call last):
  File "/app/batcontrol.py", line 1040, in <module>
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/batcontrol.py", line 113, in __init__
    self.inverter = inverter_factory.Inverter.create_inverter(config['inverter'])
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/inverter.py", line 35, in create_inverter
    inverter=FroniusWR(iv_config)
             ^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/fronius.py", line 60, in __init__
    self.set_solar_api_active(True)
  File "/app/inverter/fronius.py", line 236, in set_solar_api_active
    response_dict = json.loads(response.text)
                               ^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'text'
s6-rc: info: service legacy-services: stopping
s6-rc: info: service legacy-services successfully stopped
s6-rc: info: service legacy-cont-init: stopping
s6-rc: info: service legacy-cont-init successfully stopped
s6-rc: info: service fix-attrs: stopping
s6-rc: info: service fix-attrs successfully stopped
s6-rc: info: service s6rc-oneshot-runner: stopping
s6-rc: info: service s6rc-oneshot-runner successfully stopped

MaStr, 2025-02-10:
Ich mache mal das Update vom WR bei mir.

MaStr, 2025-02-10:
Jau, bei mir auch kaputt.

AloisKlingler, 2025-02-10:
Auszug vom changelog von 1.35.4-1:
Neue Funktionen
/ Zugriff auf der Weboberfläche der Geräte wird nun nach 5 fehlgeschlagene Login-Versuchen temporär (300 Sekunden) gesperrt.

ich habe euren login-code in node red nachgeschrieben, bei mir funktioniert es unter node red auch mit der neuen firmware von fronius.

ich habe aber (nach dem umschreiben auf nodejs) bei mir ein zwischenspeichern der cookies gemacht, dass nicht jedes mal zuerst ein 401 kommt, behandelt wird, nonce gerechnet, etc.

falls ihr das hier nicht hilft das hier eventuell, dass es bei euch auch wieder funktioniert.

https://www.photovoltaikforum.com/thread/226690-fronius-gen24-weboberfl%C3%A4che-mit-node-red-steuern/
bzw. die aktuellsten flow's:
https://www.photovoltaikforum.com/thread/226690-fronius-gen24-weboberfl%C3%A4che-mit-node-red-steuern/?postID=4162826#post4162826

der flow "401-auth-needed" macht hier die authentisierung
der flow "setupURL-checkAuth" checkt mittels timestamp die letztgültigen authentisierung nicht älter als 590 Sekunden (zumindest bei mir habe ich gesehen dass immer nach 10 Minuten der Gen24 wieder ein 401 schickt, deshalb 590 Sekunden) um immer ein gültiges cookie zu haben, bevor was gePOSTet oder geGETet wird.

LG

MaStr, 2025-02-10:
Ich glaube ich habe es gefunden. Bei mir läuft es auf dem Notebook nun.

WolfImBusch, 2025-02-10:
Hi @AloisKlingler 

tolle Flows und vielen Dank, dass Du diese bereitstellst. Ich habe mir diese gleich in Node Red importiert. Ich habe zwar batcontrol, experimentiere aber auch gerne selbst ein bisschen :-)

Ich habe gesehen, dass man das Laden der Hausbatterie sehr einfach sperren (deny-grid-charge) und wieder freigeben kann (allow-grid-charge). Ist es mit Deinem Flow auch möglich, das Laden der Hausbatterie über das Netz zu starten? Also bspw. "Lade Hausbatterie mit 3 kW bis SOC 50%".

Ich danke Dir und viele Grüße

Alex

@MaStr Sorry - hat in diesem Issue nichts verloren ..

MaStr, 2025-02-10:
> Ich glaube ich habe es gefunden. Bei mir läuft es auf dem Notebook nun.
> 
> Der Schlüssel zu den nötigen Werte aus dem authentication Header heißt jetzt „Authentication-Info“

Also der alleine ist es nicht.

AloisKlingler, 2025-02-10:
> Hi [@AloisKlingler](https://github.com/AloisKlingler)
>
> tolle Flows und vielen Dank, dass Du diese bereitstellst. Ich habe mir diese gleich in Node Red importiert. Ich habe zwar batcontrol, [...]

nimm dir Inverter-Batt-Settings ab zeile 141 als vorlage.
ich schiebe da zeitpläne der batteriesteuerung ganz oben rein und deaktiviere alle anderen drunter. anpassen der ladeleistung, des typs, und des zeitplans ist da recht übersichtlich drinnen.

und schau im photovoltaikforum in den beitrag, da passen die fragen, aber nicht hier hin. ;-)

closed 2025-02-11 by MaStr

MaStr, 2025-04-23:
@danielaussieb ich habe #155 dafür aufgemacht.