MaStr/batcontrol #128: Lock battery only when grid charging

muexxl, 2025-02-11:
Today EVCC started charging my vehicle with PV Power. As soon as the charging started the battery was locked to prevent draining. 
Unfortunately, the minimum charge power of 4.1 kW was not the entire time covered by Solar power:
When the Solar conditions got worse or the household consumption increased the available solar power dropped below the minimum possible charging level of 4.1kW. In this case EVCC keeps on charging at the minimum level for some minutes to avoid flickering. In this phases the additional required power was pulled from the grid and not from the battery. It would have been better to cover the gaps with energy from the battery. 

Enhancement proposal: 
Lock the battery only when grid charging.

MaStr, 2025-02-11:
My setup is the following:

batcontrol
always_allow_discharge_limit : 0.80
always_allow_discharge_limit : 0.75

evcc:
Battery supported charging: above 90%
  EV Start at 100%
Priorize ev change since: 85%
Priorize pv-battery until: 85%

In Addition in evcc.yaml

```
site:
  title: Haus # display name for UI
  meters:
     grid: grid # grid meter
     pv:
      - pv-fronius
      - pv-bkw
     battery:
      - battery
  residualPower: 150  # additional household usage margin
```
```
loadpoints:
  - title: WARP-Wallbox
    charger: WARP-2b5f
    mode: pv # ( now , pv , minpv , off )
    enable:
      # Enable if there is more then 1600W PV available
      threshold: -1600
      delay: 5m
    disable:
      # Disable if we import 500W for 10 minutes
      threshold: 500 # maximum import power (W)
      delay: 10m
```

This is pretty restrictive in the start during charging the PV-battery.
If it reaches 100% it simply starts and drains battery (even if the 1.4kWh minimum can't be filled with PV, intended). Eventually, the pv-battery reaches another limit, then charging stops and PV slowly fills up the pv-battery agains- then it is evening.

MaStr, 2025-02-11:
In #106 , I propose to disable the battery via batcontrol at the same values like in evcc's gui. 
As this is a subscribed topic, changes in evcc would directly affect the batcontrol behavior.