MaStr/batcontrol #136: 2 Tarif Mode

MaStr, 2025-02-26:
Native Support for low and high tariff zones, like Octopus energy.

MaStr, 2026-02-20:
Ich hab eben noch mal über das mit dem soft_price nachgedacht. Das würde sofort zum Start der low-Zone loslegen. Das wäre ungeschickt, weil dann der standby Verbrauch immer was abknapst.

Eventuell muss man doch anhand lademenge / max_charge = slots -1 vorrausschauen für den ladezeitpunkt 🤔

Edit: mit `—one-shot` kannst du alles einmal durchlaufen lassen ohne im Control Loop in eine Schleife zu hängen.. falls es hilft.

MaStr, 2026-02-21:
Ne, gleichmäßig laden über alle Stunden tut er nicht. Das passt nicht in die Logik.

Alternativ kann man einen 2 Tarif Logik Modus bauen, der dann die Eigenarten berücksichtigt.

MaStr, 2026-02-21:
> Ja, das hat noch andere Nebeneffekte. Zum Anfang einer Stunde sieht es von den Laderaten ok aus:

Lustig, die logs habe ich heute Morgen nicht so wirklich wahrgenommen.. kein mehr GitHub vor  dem ersten Kaffee.

Also das Verhalten mit der ansteigenden ChargeRate liegt an charge_rate_multiplier . Dazu gibt es einen Eintrag im Wiki.
Vielleicht sollte ich den default mal auf 1.5 anheben.

MaStr, 2026-02-23:
> Frage zur neuen Branch: verteilt er auf maximal 2 Slots, oder kann die neue Funktion auch auf 3, 4, ... n Slots verteilen? 

Sehr gute Frage! Ich hatte das extra nochmal im SourceCode nachgeschaut. Ich hatte das Feature so entwickelt, dass von nur einem Überhang auf den vorherigen Slot annimmt. Das ist für den Standardfall in der Nacht mit hoher Ladeleistung (btw: Laden mit hoher Leistung ist effizienter), wo es dann gerade so nicht mehr reicht.

Ich kann den armen Copiloten nochmal eine Runde fliegen lassen und mit geänderter Spec das umbauen lassen. ;)

OliJue, 2026-02-24:
Auf der Grafik von Octopus (https://a.storyblok.com/f/144190/300x125/410a27e5e6/octopusheat-preisgrafik.svg) sehe ich 6 Zeiten mir drei Tarifstufen. Und dann wäre da auch noch die Erwartung an das Regelergebnis. Ich kann es aber nicht testen, weil ich es nicht habe. Und ein wenig egoistisch bin ich da auch ;-) Ich bin nicht mal sicher, ob es überhaupt weitere Nutzer für den 2 Tarif Fall gibt...

Ich habe bereits zuvor für meinen Fall überlegt, dass eigentlich bessere Testmöglichkeiten fehlen: Inputreihen z.B. aus CSV für Preise, Wetter, Stromverbrauch und dann eine 24-48h Simulation (nicht real-time, also ohne die üblichen 3 Minuten Pause) mit errechnetem Akkustand. Dann könnte man gut schauen, ob die Regelwerke so funktionieren wie erwartet.