MaStr/batcontrol #137: Improve Code Structure with modern Python Packaging Standards

Julfried, 2025-02-28:
### Current State
I noticed that the project currently does not follow python packaging standards. Is there a reason for this?

As a result the project requires multiple manual steps for installation and running

```bash
# Setup
git clone https://github.com/muexxl/batcontrol.git
cd batcontrol
virtualenv venv
source venv/bin/activate
pip install -r requirements.txt

# To run:
cd /PATH/TO/batcontrol
source venv/bin/activate
python batcontrol.py
```

### Proposed Enhancement
Convert to standard Pythonackage structure:
```
repo/
├── pyproject.toml
├── src/batcontrol/
│   ├── __init__.py
│   ├── __main__.py
│   ├── forecasts/  # Current forecastsolar, forecastconsumption
│   ├── inverters/  # Current inverter modules
│   └── tariffs/    # Current dynamictariff modules
...
```

This would simplify the installation and make it more future proof as the project grows. Installation could be simplified to:
```bash
# Installation
pip install git+https://github.com/muexxl/batcontrol.git  # or from PyPI if published: pip install batcontrol
# Running
batcontrol  # or python -m batcontrol
```

I believe this would also simplify Docker deployment by allowing a cleaner Dockerfile without complex file copying steps. 

I'd be happy to implement this if there's interest, ensuring functionality remains identical.

MaStr, 2025-03-01:
This would directly affect the ha-addon , too.

Julfried, 2025-03-01:
Thanks for the feedback, I haven`t worked on ha addons before. As far as I understand HA addons run as a dockercontainer, so inside the container batcontrol could simply be installed using pip. I agree that most devs are not expercienced python developers, but I believe python packaging is quite straight forward once you are used to it while offering several benefits when dealing with larger projects.

Looking at your current approach, the HA addon repo imports batcontrol as a submodule, which can be maintained. The Dockerfile could be updated to something like:

```dockerfile
# Keep the existing submodule structure
# Then inside the container:
RUN cd /batcontrol && pip install .
```

What do you think?

Julfried, 2025-03-01:
No worries take you time to discuss this internally. If you decive to move in that direction just let me know

muexxl, 2025-03-01:
I like the idea of converting the code into a proper python package. 

The home assistant container is quite minimalistic based on alpine. 
This means we have to add pip in order to be able to install the package
e.g. add one line to the dockerfile :
`apk add py3-pip`

Thats doable. Lets go for it. 
@Julfried Thanks for contributing in advance :)

closed 2025-03-04 by MaStr

MaStr, 2025-03-06:
Thank you. I added this to the readme and will try it in the next days.

Julfried, 2025-03-06:
Yes, because batcontrol is now a package/module and not a single script anymore. So you need to run `python -m batcontrol` to indicate that you want to run it as a module.

MaStr, 2025-03-06:
I used `-c` , as sated in the "Development Mode" documentation. Yes, `-m` works.
ok, but now, the import of mqtt_api does not work anymore:

```
$ python3 -m batcontrol
2025-03-06 19:44:13 INFO [Main] Starting Batcontrol
2025-03-06 19:44:13 INFO [Batcontrol] host system time zone was not set. Setting to Europe/Berlin
2025-03-06 19:44:13 INFO [Main] MQTT Connection enabled
Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "/home/matze/projects/batcontrol/bc-git-upstream/src/batcontrol/__main__.py", line 34, in <module>
    sys.exit(main())
             ^^^^^^
  File "/home/matze/projects/batcontrol/bc-git-upstream/src/batcontrol/__main__.py", line 10, in main
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/home/matze/projects/batcontrol/bc-git-upstream/src/batcontrol/core.py", line 250, in __init__
    self.inverter.activate_mqtt(self.mqtt_api)
  File "/home/matze/projects/batcontrol/bc-git-upstream/src/batcontrol/inverter/testdriver.py", line 51, in activate_mqtt
    import mqtt_api
ModuleNotFoundError: No module named 'mqtt_api'

```

Julfried, 2025-03-06:
Nice, however there is still a small error in `pyproject.toml`. I believe the python versions should start from 3.9 and not from 3.8

MaStr, 2025-03-06:
Yes, will do this in the next commit

MaStr, 2025-03-06:
Thank you for your help!