MaStr/batcontrol #146: Logger issues after reoganizing

MaStr, 2025-03-06:
After we reorganized the project in #137 to be run via `python -m batcontrol` , the logger seems not to work correctly.

```
(.venv) matze@shrimpy:~/projects/batcontrol/bc-git-upstream$ python3 -m batcontrol
2025-03-06 20:29:44 INFO [Main] Starting Batcontrol
2025-03-06 20:29:44 INFO [Batcontrol] host system time zone was not set. Setting to Europe/Berlin
2025-03-06 20:29:44 INFO [Main] MQTT Connection enabled
2025-03-06 20:29:46 INFO [Main] evcc Connection enabled
2025-03-06 20:29:48 INFO [Main] evcc Connection ready
2025-03-06 20:29:48 WARNING [BatCtrl] always_allow_discharge_limit (0.50) is below max_charging_from_grid_limit (0.90)
2025-03-06 20:29:48 WARNING [BatCtrl] Lowering max_charging_from_grid_limit to 0.49
2025-03-06 20:29:48 DEBUG [BatCTRL] Production FCST: [  0.   0.   0.   0.   0.   0.   0.   0.   0.   0.   5. 323. 610. 828.
 885. 881. 877. 838. 768. 636. 417. 208.  20.]
2025-03-06 20:29:48 DEBUG [BatCTRL] Consumption FCST: [ 445.9  400.6  229.6  168.9  196.   163.7  166.9  166.2  182.8  284.2
  380.1  649.8  531.5  551.4 1085.6 1555.7  888.1  522.   478.9  539.6
  818.8  749.7  594.4]
2025-03-06 20:29:48 DEBUG [BatCTRL] Net Consumption FCST: [ 445.9  400.6  229.6  168.9  196.   163.7  166.9  166.2  182.8  284.2
  375.1  326.8  -78.5 -276.6  200.6  674.7   11.1 -316.  -289.1  -96.4
  401.8  541.7  574.4]
2025-03-06 20:29:48 DEBUG [BatCTRL] Prices: [0.1733 0.1498 0.1434 0.1306 0.1296 0.1296 0.1309 0.1313 0.1404 0.1604
 0.1857 0.2006 0.1691 0.1206 0.0987 0.0573 0.0338 0.028  0.0638 0.1294
 0.175  0.2033 0.2858]
2025-03-06 20:29:48 DEBUG [BatCTRL] Battery with 7589 Wh above discharge limit 5500 Wh
2025-03-06 20:29:48 INFO [Rule] Discharge allowed due to always_allow_discharge_limit
2025-03-06 20:29:48 INFO [BatCTRL] Mode: Allow Discharging
Next evaluation at 20:30:00. Sleeping for 11 seconds
```

* I'm missing a lot of debug information provided by submodules.
* `Next evaluation at..` is not with out format.

this might be THE point to align logging with Python standard, too.

Julfried, 2025-03-06:
I think this is good idea to do this, was also thinking about suggesting this. Normally in a python package you get the logger using:
```python
import logging

logging.getLogger(__name__) # Name contains the name of the current module, e.g. `inverter`
``` 

This automatically creates a new logger that streams to the "root-logger" and defines the name correct so that you can identify where this is comming from.

MaStr, 2025-03-07:
Reopened to discuss python way of logger usage.

I already did something like this here: 

https://github.com/muexxl/batcontrol/blob/main/src/batcontrol/forecastsolar/solarprognose.py

I used ˋgetChild(name)ˋ to, so I understood, get a named logger "Solarprognose". As soon as I adjust the logging format, I can automatically create the login with our  ˋ[item]ˋ tag.

commit 2344eb3f (2025-03-06): Fixes Loggin issues.

closes #146
PR #147 (2025-04-07): Streamline all the logging inside of batcontrol
Still Work in Progress

Closes #146