MaStr/batcontrol #150: Betterie lädt nicht Reserved energy ist höher als energie in akku, preis ist sehr gut

langerc, 2025-03-23:
Preis für die nächsten stunden ist unschlagbar günstig es sollte daher nachgeladen werden. Forecast ist nicht sehr gut.

stored energy ist zu wenig?

2025-03-23 10:36:19 DEBUG [Rule] Reserved Energy: **3441.7 Wh**. Usable in Battery: **0 Wh**
2025-03-23 10:36:19 DEBUG Sending PUBLISH (d0, q0, r0, m423), 'b'house/batcontrol/reserved_energy_capacity'', ... (6 bytes)
2025-03-23 10:36:19 DEBUG [Rule] Discharge forbidden. Stored usable energy 0 Wh <= Reserved energy 3441.7 Wh
2025-03-23 10:36:19 DEBUG [Rule] Discharging is NOT allowed
2025-03-23 10:36:20 DEBUG [Rule] **No additional energy required, because stored energy is sufficient**.

MaStr, 2025-03-23:
Kannst du bitte das komplette log posten mit den Zeitreihen ?

MaStr, 2025-03-23:
Ich meine das zwischen "next evaluation in...". Und deinem Auszug.

Darin sind die erwarteten Werte.

Meine Vermutung ist, das der erwartete Verbrauch für die Stunden über dem Preis-Nachlade-Limit tatsächlichen gerade noch so von der erwarteten Produktion bis zur Stunde x gedeckt wird.

langerc, 2025-03-23:
ich hoffe das passt so. zu dem 
2025-03-23 10:03:00 INFO [Main] Next evaluation at 11:06:00. Sleeping for 180 seconds
2025-03-23 10:03:59 DEBUG Sending PINGREQ
2025-03-23 10:03:59 DEBUG Received PINGRESP
2025-03-23 10:04:59 DEBUG Sending PINGREQ
2025-03-23 10:04:59 DEBUG Received PINGRESP
2025-03-23 10:05:59 DEBUG Sending PINGREQ
2025-03-23 10:05:59 DEBUG Received PINGRESP
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m74), 'b'house/batcontrol/SOC'', ... (3 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m75), 'b'house/batcontrol/stored_energy_capacity'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m76), 'b'house/batcontrol/stored_energy_capacity'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m77), 'b'house/batcontrol/always_allow_discharge_limit'', ... (4 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m78), 'b'house/batcontrol/always_allow_discharge_limit_percent'', ... (2 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m79), 'b'house/batcontrol/max_charging_from_grid_limit_percent'', ... (2 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m80), 'b'house/batcontrol/max_charging_from_grid_limit'', ... (4 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m81), 'b'house/batcontrol/min_price_difference'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m82), 'b'house/batcontrol/min_price_difference_rel'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m83), 'b'house/batcontrol/evaluation_intervall'', ... (3 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m84), 'b'house/batcontrol/last_evaluation'', ... (10 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m85), 'b'house/batcontrol/discharge_blocked'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m86), 'b'house/batcontrol/inverters/0/SOC'', ... (3 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m87), 'b'house/batcontrol/inverters/0/stored_energy'', ... (17 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m88), 'b'house/batcontrol/inverters/0/free_capacity'', ... (18 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m89), 'b'house/batcontrol/inverters/0/max_capacity'', ... (7 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m90), 'b'house/batcontrol/inverters/0/usable_capacity'', ... (18 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m91), 'b'house/batcontrol/inverters/0/max_grid_charge_rate'', ... (4 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m92), 'b'house/batcontrol/inverters/0/max_pv_charge_rate'', ... (4 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m93), 'b'house/batcontrol/inverters/0/min_soc'', ... (1 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m94), 'b'house/batcontrol/inverters/0/max_soc'', ... (3 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m95), 'b'house/batcontrol/inverters/0/capacity'', ... (7 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m96), 'b'house/batcontrol/max_energy_capacity'', ... (7 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m97), 'b'house/batcontrol/always_allow_discharge_limit_capacity'', ... (7 bytes)
2025-03-23 10:06:00 DEBUG [FC Cons] predicting consumption: [1965.1 1248.1 1355.   651.1  496.7  912.  1464.3  986.  1136.1  942.9
  703.   498.5  425. ]
2025-03-23 10:06:00 DEBUG [BatCTRL] Production FCST: [1675. 1743. 1339. 1121.  889.  605.  260.   21.    0.    0.    0.    0.
    0.]
2025-03-23 10:06:00 DEBUG [BatCTRL] Consumption FCST: [1965.1 1248.1 1355.   651.1  496.7  912.  1464.3  986.  1136.1  942.9
  703.   498.5  425. ]
2025-03-23 10:06:00 DEBUG [BatCTRL] Net Consumption FCST: [ 290.1 -494.9   16.  -469.9 -392.3  307.  1204.3  965.  1136.1  942.9
  703.   498.5  425. ]
2025-03-23 10:06:00 DEBUG [BatCTRL] Prices: [0.0299 0.0347 0.0242 0.0322 0.0713 0.128  0.1776 0.1961 0.1955 0.1934
 0.1802 0.1743 0.1592]
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m98), 'b'house/batcontrol/FCST/production'', ... (939 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m99), 'b'house/batcontrol/FCST/consumption'', ... (1108 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m100), 'b'house/batcontrol/FCST/net_consumption'', ... (1109 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m101), 'b'house/batcontrol/FCST/prices'', ... (958 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m102), 'b'house/batcontrol/stored_usable_energy_capacity'', ... (3 bytes)
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m103), 'b'house/batcontrol/min_dynamic_price_difference'', ... (5 bytes)
2025-03-23 10:06:00 DEBUG [Rule] Evaluating next 13 hours until 23:59
2025-03-23 10:06:00 DEBUG [Rule] Reserved Energy will be used in the next hours: [1, 3, 4, 5, 6, 7, 8, 9, 10, 11, 12]
2025-03-23 10:06:00 DEBUG [Rule] Reserved Energy: 4824.7 Wh. Usable in Battery: 0.0 Wh
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m104), 'b'house/batcontrol/reserved_energy_capacity'', ... (6 bytes)
2025-03-23 10:06:00 DEBUG [Rule] Discharge forbidden. Stored usable energy 0.0 Wh <= Reserved energy 4824.7 Wh
2025-03-23 10:06:00 DEBUG [Rule] Discharging is NOT allowed
2025-03-23 10:06:00 DEBUG [Rule] No additional energy required, because stored energy is sufficient.
2025-03-23 10:06:00 DEBUG [BatCTRL] Charging allowed: True
2025-03-23 10:06:00 DEBUG [Rule] Charging is allowed, because SOC is below 90%
2025-03-23 10:06:00 DEBUG [Rule] No additional energy required or possible price found.
2025-03-23 10:06:00 INFO [BatCTRL] Mode: Avoid Discharging
2025-03-23 10:06:00 INFO [Inverter] Login successful <Response [200]>
2025-03-23 10:06:00 DEBUG Sending PUBLISH (d0, q0, r0, m105), 'b'house/batcontrol/mode'', ... (1 bytes)
2025-03-23 10:06:00 INFO [Main] Next evaluation at 11:09:00. Sleeping for 180 seconds

MaStr, 2025-03-23:
Ist der Auszug aktuell?
Wenn ja: korrigiere bitte deine Zeitzone. Wenn die nicht stimmt, kann es zu Unstimmigkeiten kommen. Der Auszug zeigt 10 Uhr an. Aktuell ist 12?

Was hast du als min_price_diff? Er sagt, dass er keine Zeit mit passenden preisziel gefunden hat:

2025-03-23 10:06:00 DEBUG [Rule] No additional energy required or possible price found.

Ansonsten hast du ein klein wenig Produktion heute, das wird er vermutlich dann auch aufsparen für die teuerste Stunde.

langerc, 2025-03-23:
der auszug ist von 10h - ich habe dann versucht mit charge_rate/set eine ladung zu erzwingen - siehe anderes ticket ich habe hin und wieder tage wo sehr viel strom benögit wird (brautag) heute wird generell viel strom benötigt und er lädt nicht.

hat aber keine eile du brauchst dich nicht am sonntag stressen lassen  ! in jedem fall vielen dank für das geniale tool!!

MaStr, 2025-03-23:
Ok, trotzdem wäre ein Auszug der config.yml nicht schlecht um mal zu schauen, ob die Grenzwerte ordentlich verwendet wurden.

langerc, 2025-03-23:
sorry hab ich falsch verstanden.

timezone: Europe/Berlin #your time zone. not optional.
loglevel: debug
logfile_enabled: true
max_logfile_size: 100 #kB
logfile_path: logs/batcontrol.log
battery_control:
  min_price_difference: 0.05 # minimum price difference in Euro to justify charging your battery
  always_allow_discharge_limit: 0.65 # 0.00 to 1.00 above this SOC limit using energy from the battery is always allowed
  max_charging_from_grid_limit: 0.90 # 0.00 to 1.00 charging from the grid is only allowed until this SOC limit
inverter:
  type: fronius_gen24 #currently only fronius_gen24 supported
  address: ..... # the local IP of your inverter. needs to be reachable from the machine that runs batcontrol
  user: customer #customer or technician lowercase only!!
  password: ...... #
  max_grid_charge_rate: 8000 # Watt
  max_pv_charge_rate : 8000 # Watt
utility:
  type: awattar_at # [tibber, awattar_at, awattar_de, evcc]
  apikey: YOUR-PASSWORD 
  vat: 0.20 # only required for awattar
  fees: 0.015 # only required for awattar
  markup: 0.03 # only required for awattar
  url: http://evcc.local:7070/api/tariff/grid # only required for evcc
mqtt: ...

pvinstallations:
  - name: private
    lat: 48.30639
    lon: 14.28611
    declination: 15 
    azimuth: 0 
    kWp: 5.0
    api: ....
  - name: sidewall #name
    lat: 48.30639
    lon: 14.28611
    declination: 15
    azimuth: 0
    kWp: 2.6
    api: .....

MaStr, 2025-03-23:
Ok, danke dir.

Zwei Anmerkungen:

- Du musst das allow_always_discharge Limit über den charge_limit setzen. Das ist die Prüfreihenfolge. (Ab 0.5 wird batcontrol das mit einer Warnung quittieren und selbst ändern).
- im Standard (bei dir) lädt batcontrol **nur** zum niedrigsten Preisspunkt auf. Das war bei dir noch nicht erreicht. Wenn du jetzt einen sehr großen Speicher hast, der mehr als eine Stunde braucht zum Laden, kannst du die Parameter tunen.

https://github.com/muexxl/batcontrol/wiki/battery_control_expert#adjust-charging-pricepoint

Das bedarf mindestens Version 0.4 . Mit der Anpassung wird nicht nur am niedrigsten Punkt, sondern eventuell etwas früher gestartet mit dem nachladen.

MaStr, 2025-03-31:
Kann ich hier zumachen?

closed 2025-03-31 by langerc