MaStr/batcontrol #151: Ziel ladung in % per mqtt setzen

langerc, 2025-03-23:
Es steht ein geplanter großverbrauch bevor der ausserhalb der norm ist. Statt normalerweise 20 werden 60kw benötigt. Gut wäre es auch noch wenn die batterie am ende voll ist weil die energie günstig ist.

charge_rate/set - funktioniert, setzt sich jedoch nach ein paar minuten zurück

optimal wäre charge_rate/set und target_charge = 80%

langerc, 2025-03-23:
Für den Fall das jemand so ein script sucht:
linux mqtt client
apt-get install mosquitto-clients

#!/bin/bash

# MQTT server details
MQTT_SERVER="" # Replace with your MQTT ip
MQTT_TOPIC="house/batcontrol/charge_rate/set"
MQTT_PAYLOAD="4000"

# MQTT credentials
MQTT_USERNAME=""  # Replace with your MQTT username
MQTT_PASSWORD=""  # Replace with your MQTT password

# Duration in seconds (3 hours = 10800 seconds)
DURATION=10800

# Interval in seconds (1 minute = 60 seconds)
INTERVAL=60

# Calculate the number of iterations
ITERATIONS=$((DURATION / INTERVAL))

# Loop to send the MQTT message
for ((i=0; i<ITERATIONS; i++))
do
    mosquitto_pub -h $MQTT_SERVER -t $MQTT_TOPIC -m $MQTT_PAYLOAD -u $MQTT_USERNAME -P $MQTT_PASSWORD
    sleep $INTERVAL
done

MaStr, 2025-03-23:
Ja genau.
Ich hatte mir einfach einen cronjob eingerichtet, der das Kommando an einem bestimmten Tag zu einer bestimmten Zeiten minütlich ausführt.