MaStr/batcontrol #155: Fehler nach firmwareupdate von Fronius auf 1.36.5-1

MaStr, 2025-04-23:
bin mal wieder vorgeprescht mit dem nächsten firmwareupdate von Fronius auf 1.36.5-1

nun folgender fehler:

2025-04-23 16:36:00 INFO [Inverter] loading module 
Traceback (most recent call last):
  File "/app/batcontrol.py", line 1071, in <module>
    bc = Batcontrol(CONFIGFILE)
         ^^^^^^^^^^^^^^^^^^^^^^
  File "/app/batcontrol.py", line 120, in __init__
    self.inverter = inverter_factory.Inverter.create_inverter(
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/inverter.py", line 35, in create_inverter
    inverter=FroniusWR(iv_config)
             ^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/fronius.py", line 74, in __init__
    self.previous_battery_config = self.get_battery_config()
                                   ^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/fronius.py", line 143, in get_battery_config
    response = self.send_request(path, auth=True)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/app/inverter/fronius.py", line 483, in send_request
    raise RuntimeError(
RuntimeError: [Inverter] Request 0 failed with 404-Not Found. 
	 path:/config/batteries, 
	params:None 
	headers {} 
	nonce 0 
	payload 

_Ursprünglich gepostet von @danielaussieb in https://github.com/muexxl/batcontrol/issues/125#issuecomment-2824750605_

Brandy43, 2025-05-08:
Gibt es schon eine Lösung?

MaStr, 2025-05-08:
Noch keine Zeit gehabt.. zumal die Firmware zwischendurch zurückgezogen war.

Ich versuche am Sonntag da mal eine Lösung zu implementieren.

Brandy43, 2025-05-08:
Super danke inzwischen gibt es die Firmware ROW 1.36.6-1

MaStr, 2025-05-08:
Danke, Ja, habe ich gesehen.. ist der fix für den Grund warum 1.35 zurückgezogen wurde.

MaStr, 2025-05-10:
Die neuen API Endpunkte sind auf den Hauptbranch.

closed 2025-05-10 by MaStr

muexxl, 2025-05-11:
Super. Sollten wir mal wieder ein Bugfixrelease machen ?

MaStr, 2025-05-11:
@muexxl soll Ich noch ein 0.4.x Release bauen? Dann backporte ich das erstmal, sodass wir das auch schneller rausstellen können

Brandy43, 2025-05-12:
Kommt dann bald eine neue version für HA Addon raus?

MaStr, 2025-05-12:
@johannesghd jap, soweit erledigt. Kein Problem.

Brandy43, 2025-05-12:
Habe gerade mal den Main auf meiner Synology Nas laufen! Funktioniert wieder 1a! Danke an euch!

MaStr, 2025-05-12:
> Habe gerade mal den Main auf meiner Synology Nas laufen! Funktioniert wieder 1a! Danke an euch!
@Brandy43 
Bitte, wird auch bald ein ordentliches Release geben. (Inklusive ha addon).
Wir haben nur größere Änderungen in 0.5, was initial etwas mehr Arbeit benötigt.